majianjia/nnom #13: Rethiking the affect of deleting qformat()

majianjia, 2019-03-13:
Currently, the Q format is handled by the scripts. 
The nnom cannot really know what is the q format of the layers outputs but their output shifting. 

This is fine with most of the layers, such as conv, dense, and relu. 
However, the activations such as sigmoid and tanh must know the current Q format because they are doing arithmetic base on the real number. There is a "num int bit" argument, which should be the m in Qmn format. 

I am still thinking of the solution. 

@parai 
 Do you have any suggestion?
Thanks

parai, 2019-03-13:
Oh, Let me think and get back to you.

parai, 2019-03-14:
I didn't think up a good way that could make nnom still looks good and pretty if not using qformat for each layer.
The ugly way is to add a parameter to TanH/Sigmoid layer and this is very easy to implement this by scripts, below code could be a demo:
```c
layer[...] = TanH(INPUT_LAYER_Q);
```
But this is really ugly as I think.

But if restore to using Q Format for each layer, this is somehow not necessary for most of the layers.

So somehow I prefer the ugly way.

majianjia, 2019-04-11:
Thanks, @parai 
Its time to fixed this issue
I am now planning to do it like the way you did. Together with Mult() and Add() layer.

majianjia, 2019-04-11:
Issue solve after https://github.com/majianjia/nnom/pull/30

closed 2019-04-11 by majianjia