majianjia/nnom #28: Do we have a plan to make mcu train its model self?

licunhao, 2019-04-11:
Now I know nnom is a way to do guess on mcu.
We should train our model on keras.Then use script to generate model c files to our project.
Finally our mcu run the model and guess the result.

It is not a fully edge AI way.Is there a plan to make mcu train and create its model self?
I am looking forward to you reply.

majianjia, 2019-04-11:
Your understanding is correct. Currently, training must be done in a PC with Keras. 

Training requires higher precision, especially when calculating loss and updating weights, which in my point of view has to be done with floating point data. NNoM currently runs on 8 bit only and plan to support 16bit later. Supporting floating-point is beyond my target to run NN on MCU, however, PR welcomed. 

If there is a method for training the fixed-point model efficiently, I will try to make it work on NNoM. 
In addition, training requires much more memory which might be a burden for MCU. 

I know there are some libs support online training with MLP layers only with floating-point arithmetic, which might be an alternative to the fixed-point lib if your model is simple enough.

licunhao, 2019-04-12:
I got that.
Thank you.

I will consider your ideal.

closed 2019-04-12 by licunhao