majianjia/nnom #29: Wonderful Project! Just mention a relevant project

dustinjoe, 2019-04-11:
Hi, this is a wonderful project.
Actually I just noticed that Tensorflow team just released a MCU version that can run TFLite model on microcontrollers.
[https://www.sparkfun.com/products/15170](url)

majianjia, 2019-04-11:
Thanks for the info, 
I have checked the TFlite demo video with Sparkfun boards some times ago. Keyword spotting is a great demo to attract people :-) 
So I built one KWS example for NNoM just now. This example is now available under `examples/keyword_spotting`. Docs are preparing, by the meantime, please raise issues if you have any problem about it, I will be happy to help. 

The different between TFlite and NNoM is obvious. TFLite is developed from a comprehensive functional lib which might be bulky for MCUs, while NNoM starts from very low-level functions but aims to provide a user-friendly interface.

licunhao, 2019-04-12:
```

- Micro Speech Example
- This examples shows how you can use TensorFlow Lite to run a 20 kilobyte neural network model to recognize keywords in speech. It's designed to run on systems with very small amounts of memory such as microcontrollers and DSPs. The code itself also has a small footprint (for example around 22 kilobytes on a Cortex M3) and only uses about 10 kilobytes of RAM for working memory, so it's able to run on systems like an STM32F103 with only 20 kilobytes of total SRAM and 64 kilobytes of Flash.

```
above is picked https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/experimental/micro/examples/micro_speech#creating-your-own-model

It shows run model cost 20KB ram and 22KB flash.
I think it is not really bulky for MCUs.
Now I am doing the investigating for TFLite on MCU.

majianjia, 2019-04-12:
@licunhao 
Thanks for the info :-) 
Please do check the TFlite for MCU. I am happy to discuss about it.

closed 2019-04-13 by dustinjoe