majianjia/nnom #42: Tensor objects are only iterable when eager execution is enabled.

Alan-Turing-Ko, 2019-06-07:
Wonderful project!
I have one issue.
When I generate model, I got following error.

Traceback (most recent call last):
  File "keras_to_nnom.py", line 18, in <module>
    generate_model(model, np.ones(shape=(1, 128, 128, 1)), name=weights)
  File "/mnt/d/NUC970/Proj/nnom-master/scripts/nnom_utils.py", line 546, in generate_model
    inps = [input.name.replace(':','/').split('/')[0] for input in layer.input]
  File "/mnt/d/NUC970/Proj/nnom-master/scripts/nnom_utils.py", line 546, in <listcomp>
    inps = [input.name.replace(':','/').split('/')[0] for input in layer.input]
  File "/home/turing/.local/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 442, in __iter__
    "Tensor objects are only iterable when eager execution is "
TypeError: Tensor objects are only iterable when eager execution is enabled. To iterate over this tensor use tf.map_fn.

I am using tensorflow 1.13.1
Hope to any suggentions.

Alan-Turing-Ko, 2019-06-07:
I found the problem.
Above error come from ZeroPadding2D layer.
Can you possible add ZeroPadding2D support?
I tried convert my caffe model, and because in caffe, VALID pad is used mostly.
That makes ZeroPadding2D layer when convert to keras.
I think ZeroPadding2D support will be very necessary to whom trys to deploy caffe model using your great framework.
And if possible, flatten layer.
Thank you.

majianjia, 2019-06-07:
Hi @Alan-Turing-Ko,
Thanks for the suggestion. ZeroPadding2D sounds necessary in your cases. 
I am going to implement ZeroPadding2D now. I will let you know when it is done and if you can please test after. 
What do you think of Cropping1/2D? They have a similar interface to zero paddings but I am not sure whether they are necessary as well.

majianjia, 2019-06-07:
Hi @Alan-Turing-Ko 
ZeroPadding1/2D is working now. Please check the "dev branch". 
I appreciate if you can help with the testing. 

Below is the testing log with MNIST. The efficiency is not evaluated but it looks ok, takes 90us and 60us on an STM32L476 @ 150MHz. 
~~~
 \ | /
- RT -     Thread Operating System
 / | \     4.0.0 build Jun  7 2019
 2006 - 2018 Copyright by rt-thread team
RTT Control Block Detection Address is 0x20000b64

Start compiling model...
Layer(#)         Activation    output shape    ops(MAC)   mem(in, out, buf)      mem blk lifetime
-------------------------------------------------------------------------------------------------
#1   Input      -          - (  28,  28,   1)          (   784,   784,     0)    1 - - -  - - - - 
#2   Conv2D     - ReLU     - (  28,  28,  12)      84k (   784,  9408,    36)    1 1 1 -  - - - - 
#3   MaxPool    -          - (  14,  14,  12)          (  9408,  2352,     0)    1 1 1 -  - - - - 
#4   ZeroPad    -          - (  17,  21,  12)          (  2352,  4284,     0)    1 1 - -  - - - - 
#5   Conv2D     - ReLU     - (  17,  21,  24)     925k (  4284,  8568,   432)    1 1 1 -  - - - - 
#6   MaxPool    -          - (   5,   6,  24)          (  8568,   720,     0)    1 1 1 -  - - - - 
#7   ZeroPad    -          - (   8,  13,  24)          (   720,  2496,     0)    1 1 - -  - - - - 
#8   Conv2D     - ReLU     - (   8,  13,  24)     539k (  2496,  2496,   864)    1 1 1 -  - - - - 
#9   UpSample   -          - (  16,  26,  24)          (  2496,  9984,     0)    1 - 1 -  - - - - 
#10  Conv2D     - ReLU     - (  16,  26,  48)    4.31M (  9984, 19968,   864)    1 1 1 -  - - - - 
#11  MaxPool    -          - (   8,  13,  48)          ( 19968,  4992,     0)    1 1 1 -  - - - - 
#12  Dense      - ReLU     - (  64,   1,   1)     319k (  4992,    64,  9984)    1 1 1 -  - - - - 
#13  Dense      -          - (  10,   1,   1)      640 (    64,    10,   128)    1 1 1 -  - - - - 
#14  Softmax    -          - (  10,   1,   1)          (    10,    10,     0)    1 1 - -  - - - - 
#15  Output     -          - (  10,   1,   1)          (    10,    10,     0)    1 - - -  - - - - 
-------------------------------------------------------------------------------------------------
Memory cost by each block:
 blk_0:9984  blk_1:4992  blk_2:19968  blk_3:0  blk_4:0  blk_5:0  blk_6:0  blk_7:0  
 Total memory cost by network buffers: 34944 bytes
Compling done in 180 ms
msh >nn
nn_stat
msh >nn_stat

Print running stat..
Layer(#)        -   Time(us)     ops(MACs)   ops/us 
--------------------------------------------------------
#1        Input -        11                  
#2       Conv2D -      5848          84k     14.47
#3      MaxPool -       698                  
#4      ZeroPad -        90                  
#5       Conv2D -     15227         925k     60.76
#6      MaxPool -      1686                  
#7      ZeroPad -        60                  
#8       Conv2D -      8412         539k     64.09
#9     UpSample -       223                  
#10      Conv2D -     64503        4.31M     66.86
#11     MaxPool -      4066                  
#12       Dense -      5534         319k     57.73
#13       Dense -        16          640     40.00
#14     Softmax -         4                  
#15      Output -         1                  

Summary:
Total ops (MAC): 6182368(6.18M)
Prediction time :106379us
Efficiency 58.11 ops/us
NNOM: Total Mem: 37036
msh >
~~~

Alan-Turing-Ko, 2019-06-12:
Thank you for your support.
Now it works well.
Thank you again for this great framework.

closed 2019-06-12 by Alan-Turing-Ko