majianjia/nnom #46: Compilation Error With Layer Output RSHIFT

Kelvin-Chan, 2019-06-14:
Hello, great library you have here! I was looking for something exactly like this! I am currently using v0.2.0_beta.

I am trying to convert a Keras model to compile on ARM Cortex M7 with CMSIS NN optimization enabled, but I am running to this issue during compilation:

```
../source/weight.h:76:2: error: #error DENSE_2_OUTPUT_RSHIFT must be bigger than 0
```
Looking at the generated weight file, I find the lines:

```c
#define DENSE_1_OUTPUT_SHIFT 0
// ......
#define DENSE_2_KERNEL_0_SHIFT (6)
// ......
#define DENSE_2_OUTPUT_SHIFT 7
// ......
#define DENSE_2_OUTPUT_RSHIFT (DENSE_1_OUTPUT_SHIFT+DENSE_2_KERNEL_0_SHIFT-DENSE_2_OUTPUT_SHIFT)
```
Which would obviously result in -1...

Additionally, while running the utility script to convert my Keras model, I also ran into the following issue in line 582 of [nnom_utils.py](https://github.com/majianjia/nnom/blob/master/scripts/nnom_utils.py) as it tried to generate the last Dense layer with softmax activation.
https://github.com/majianjia/nnom/blob/3f245721b14221a5e3c3a8e8760666bed94eafbf/scripts/nnom_utils.py#L578-L582

```python
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-6-f285063f1d0f> in <module>()
      6 print(model.layers)
      7 
----> 8 generate_model(model, x, name='weights.h')

/home/kelvinchan/kick-sensor-cnn/nnom/scripts/nnom_utils.py in generate_model(model, x_test, name)
    580             fp.write('\tlayer[%s] = model.hook(Output(shape(%s,1,1), nnom_output_data), layer[%s]);\n'%(id+1, layer.input.shape[1], id))
    581         else:
--> 582             fp.write('\tlayer[%s] = model.hook(Output(shape%s, nnom_output_data), layer[%s]);\n'%(id+1, layer.shape[1:], id))
    583         fp.write('\tmodel_compile(&model, layer[0], layer[%s]);\n'%(id+1))
    584         if(ID>32):

AttributeError: 'Dense' object has no attribute 'shape'
```
By changing `layer.shape[1:]` to `layer.input.shape[1:]`, I got it to successfully generate `weights.h`, but I'm not sure if it is the reason why I encounter the first issue.

My Keras model summary:
```
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
conv1d_1 (Conv1D)            (None, 998, 64)           448       
_________________________________________________________________
conv1d_2 (Conv1D)            (None, 996, 64)           12352     
_________________________________________________________________
dropout_1 (Dropout)          (None, 996, 64)           0         
_________________________________________________________________
max_pooling1d_1 (MaxPooling1 (None, 498, 64)           0         
_________________________________________________________________
flatten_1 (Flatten)          (None, 31872)             0         
_________________________________________________________________
dense_1 (Dense)              (None, 100)               3187300   
_________________________________________________________________
dense_2 (Dense)              (None, 8)                 808       
=================================================================
Total params: 3,200,908
Trainable params: 0
Non-trainable params: 3,200,908
_________________________________________________________________
```

majianjia, 2019-06-14:
Hi @Kelvin-Chan,

**The first** problem is due to the CMSIS-NN only support right shifts in the dense layer, same as convolution layer. So we add a macro to check in case we get negative right shifts. 

There is no good solution to this currently. You can adjust your model's hyper-parameters and check the log output when you call `generate_model()`. It should print the shifts for every layer. 
BTW, your model seems to be too large for MCU. The parameters for dense_1 is too much which I think can be the reason for this. If you can reduce the size of the output of conv1d_2, it might help. 

**The second** error is because the script was not designed to handle the dense layer as the output layer correctly.

The solution for this can be either:
1. use a softmax layer as the final output. OR 
2. temporary use the below code to replace the code at line 578~582 in your script. 

~~~python
if('softmax' in layer.name 
    or ('activation' in layer.name and layer.get_config()['activation'] == 'softmax')): 
     fp.write('\tlayer[%s] = model.hook(Output(shape(%s,1,1), nnom_output_data), layer[%s]);\n'%(id+1, layer.input.shape[1], id)) 
 else: 
     fp.write('\tlayer[%s] = model.hook(Output(shape(%s,1,1), nnom_output_data), layer[%s]);\n'%(id+1, layer.input.shape[1], id)) 
~~~

The solution 2 is not the solution for all models. I will try to fix this issue later.

Kelvin-Chan, 2019-06-14:
Hi @majianjia ,

Thank you for the suggestion! I was able to successfully generate the model after redefining the Keras model from Sequential to Functional and adding in the extra softmax layer. I still had to comment out the header error for `DENSE_2_OUTPUT_LSHIFT < 0` though during compilation for the board, but since you mentioned that CMSIS-NN only supports right shifts, I don't think it'll be an issue.

I am currently using [IMXRT1050-EVKB](https://www.nxp.com/support/developer-resources/evaluation-and-development-boards/i.mx-evaluation-and-development-boards/i.mx-rt1050-evaluation-kit:MIMXRT1050-EVK) right now. which has additional memory locations available. I was able to compile and load the whole program with the network onto the board, so that seems promising.

I will try to test out the compiled model on the test dataset (I believe there were Python utility functions for converting test inputs to headers?) and report the performance.

closed 2019-06-14 by Kelvin-Chan

majianjia, 2019-06-15:
Hi @Kelvin-Chan,
For validating, you can convert the test data to a binary file using this python api `generate_test_bin()`
https://majianjia.github.io/nnom/api_nnom_utils/#generate_test_bin
It will generate a binary file with label and test data using a very simple structure. 

In a few examples, it uses Y-modem to download the file on to MCU for validating. However, you can always change the file source to your local storage, such as the file system in your development board. 

If you want to generate a few data into a header, you can check `mnist_simple` example https://github.com/majianjia/nnom/tree/master/examples/mnist-simple
especially this method https://github.com/majianjia/nnom/blob/dc21084912455a28d50ffb4a5a76b14bb714d332/examples/mnist-simple/model/mnist_simple.py#L23
Then in the MCU, use one of these apis to run your model and get the predicted label. 
https://majianjia.github.io/nnom/api_evaluation/

To validate the performance, please you can use this API `model_stat()`. It will report the last runtime performance. 
https://majianjia.github.io/nnom/api_evaluation/#model_stat

You to use model_stat(), you will need to provide a microsecond time source. Which is a simple function return the microsecond timestamp of the system. This is normally done by a hardware timer configured to 1us step. So you need to manually initialise and start it before you run the model.  

Please check this [uci-inception example](https://github.com/majianjia/nnom/tree/master/examples/uci-inception). 
Here is an example to provide ms and us timestamps for performance evaluation. you need to modified the `nnom_port.h`. Documentation: https://majianjia.github.io/nnom/Porting_and_Optimisation_Guide/
~~~
// runtime & debuges
#define nnom_us_get()		us_timer_get()
#define nnom_ms_get()		systick_get()
#define NNOM_LOG(...)    printf(__VA_ARGS__)
~~~