majianjia/nnom #49: 'Dense' object has no attribute 'shape'

miaowrx, 2019-06-19:
When I try a simple code like than:

```python
import numpy as np
import keras
import nnom_utils

x_train = np.array([[0, 0],
                    [0, 1],
                    [1, 0],
                    [1, 1]])

y_label = np.array([0, 1, 1, 0])

model = keras.Sequential()
model.add(keras.layers.Dense(8, input_shape=(2,), activation='relu'))
model.add(keras.layers.Dense(8, activation='relu'))
model.add(keras.layers.Dense(1, activation='sigmoid'))

model.compile(optimizer='adam', loss='binary_crossentropy')

model.fit(x_train, y_label, batch_size=1, epochs=100)

nnom_utils.generate_model(model, x_train * 0.9)
```

I got an error:

```
AttributeError: 'Dense' object has no attribute 'shape'
```

majianjia, 2019-06-19:
Hi @miaowrx ,

This is an issue related to https://github.com/majianjia/nnom/issues/46
It is due to the current script cannot handle the dense layer as the final output. 

Solutions:
1. I am currently fixing this issue. The new script in dev branch might works. Or 
2.  you can add a softmax as your final output to your model. 

Please tell me if you have any further problem. 
Thanks

miaowrx, 2019-06-19:
thanks for your suggestion, I try it again.

closed 2019-06-19 by miaowrx

franchesoni, 2020-11-03:
What if I want a regression output? Softmax does not seem right.

majianjia, 2020-11-04:
This should already be fixed. You could use dense as the final output now. 
Regression output is demonstrated in the speech noise suppression example https://github.com/majianjia/nnom/tree/master/examples/rnn-denoise
The output of them is the regression of  frequency band gains in speech.