majianjia/nnom #57: About Batchnorm

nicktao9, 2019-07-02:
thanks for your contributation,The question of me is: when the nets inference,why we all do not use batchnorm?whether your all nets-mcu  project do not use batchnorm ?
I just do a project,my nets use batchnorm,but when I move to MCU，I find much Calculated amount added,what can I do?thank you

majianjia, 2019-07-02:
Hi @nicktao9 
Sorry but can you explain "calculated amount added" a bit more?
BN layer in here should be fused to the weighs of last convolution layer already. Therefore it should reduce the calculation.

nicktao9, 2019-07-02:
> Hi @nicktao9
> Sorry but can you explain "calculated amount added" a bit more?
> BN layer in here should be fused to the weighs of last convolution layer already. Therefore it should reduce the calculation.
I am calculating the output through the BN layer in the MCU.not fused to the weights,I have find the fused method from your script,thank you .best wishes