majianjia/nnom #61: Strange Predictions with Sigmoid Activation

Kelvin-Chan, 2019-07-04:
Hello again,

I have been trying to implement a small 1D CNN binary classifier with sigmoid activation at the end. When I copy my test inputs to `nnom_input_data` and run `nnom_predict` with the compiled model, `prob` always output to 1 for some reason, even though I verified the model prediction results with the same test data on Keras.

Here is the summary of the network from Keras:

```
Layer (type)                 Output Shape              Param #   
=================================================================
input_1 (InputLayer)         (None, 1000, 2)           0         
_________________________________________________________________
conv1d_1 (Conv1D)            (None, 991, 4)            84        
_________________________________________________________________
max_pooling1d_1 (MaxPooling1 (None, 495, 4)            0         
_________________________________________________________________
re_lu_1 (ReLU)               (None, 495, 4)            0         
_________________________________________________________________
conv1d_2 (Conv1D)            (None, 489, 8)            232       
_________________________________________________________________
max_pooling1d_2 (MaxPooling1 (None, 244, 8)            0         
_________________________________________________________________
re_lu_2 (ReLU)               (None, 244, 8)            0         
_________________________________________________________________
conv1d_3 (Conv1D)            (None, 240, 16)           656       
_________________________________________________________________
max_pooling1d_3 (MaxPooling1 (None, 120, 16)           0         
_________________________________________________________________
re_lu_3 (ReLU)               (None, 120, 16)           0         
_________________________________________________________________
global_max_pooling1d_1 (Glob (None, 16)                0         
_________________________________________________________________
dense_1 (Dense)              (None, 50)                850       
_________________________________________________________________
re_lu_4 (ReLU)               (None, 50)                0         
_________________________________________________________________
dense_2 (Dense)              (None, 1)                 51        
_________________________________________________________________
activation_1 (Activation)    (None, 1)                 0         
=================================================================
Total params: 1,873
Trainable params: 1,873
Non-trainable params: 0
```

I am following the procedure from the [mnist-simple](https://github.com/majianjia/nnom/blob/master/examples/mnist-simple/mcu/main.c) example:

```c
int8_t data[1000][2000];
uint32_t predict_label;
float prob = 0;

// load data from binary file

model = nnom_model_create();
memcpy(nnom_input_data, data[i], sizeof(nnom_input_data));

nnom_predict(model, &predict_label, &prob);

printf("Probability: %.2f \r\n", prob);
```

Currently using v0.2.0 in the MCU code.
If anyone has any insights or advice, that'll be much appreciated.

majianjia, 2019-07-05:
Hi @Kelvin-Chan 
Another developer encountered the same problem. 
His model is similar to yours with one neural output and is using sigmoid as the final activation. 

For your issue specifically, the `nnom_predict()` cannot handle one neural output properly. 
I am fixing this problem.

majianjia, 2019-07-05:
Hi, @Kelvin-Chan 
This issue should be fixed. It is caused by a bug in a signed/unsigned argument indicating how many integers bits in the input data. When it is larger than 7, the integer bit can be negative. It was passed as an unsigned value.  This bug affects both tanh and sigmoid. Now has been fixed. 

Please download the new master branch for this fix.

Additionally. 
`nnom_predict()` can now work with a single neural.  It will go with this new routine for a single neural as output. 
prob = neural / 127
If prob < 0.5, label = 0. 
If prob >=0.5, label = 1.

majianjia, 2019-07-05:
@Kelvin-Chan 
BTW, the current script does not support activations attached to layer, such as
`Dense(30, activation ='sigmoid')`
Please replace them like this
~~~
Dense(32)
Sigmoid()
~~~

Kelvin-Chan, 2019-07-05:
@majianjia 
Thank you, that fixed the prediction issue!

Yes, I noticed that too while converting a pretrained model, ended up having to redefine the network with explicit activation layers and reload the weights.

Are there any ongoing effort to support conversion of attached activation in the script? If not, I would like to take a stab at it and contribute to this cool project. 😄

closed 2019-07-05 by Kelvin-Chan

majianjia, 2019-07-05:
Hi @Kelvin-Chan 
I have my initial trial for supporting these implicit activations but did not work well. You are welcome to contribute everything on NNoM. 
My plan on the scripts is to completely rebuild them. The current `nnom_utils.py` was for temporary use for my initial experiment with NNoM. Every method in it is too long and not well organised. 

The next plan is to get support for [TensorLayer](https://github.com/tensorlayer/tensorlayer) by using [Maix-EMC](https://github.com/sipeed/Maix-EMC) or by our new scripts.