majianjia/nnom #67: 关于CMSIS-NN

Be997398715, 2019-09-06:
我使用了CMSIS-NN，在 port.h中也开启了，并且版本都是符合的，但是在实验中，使用了NN却没有提升太多的速度，比如不使用预测需要230 tick,使用后变为190 tick, 而在我自己写的神经网络中，不使用NN预测需要70 tick，使用后变为51 tick。 使用的硬件为stm32f407zet6，环境都是搭好的，rt-thread中1s 1000tick

majianjia, 2019-09-06:
请检查你是否在工程中开启 `ARM_MATH_CM4,` __FPU_PRESENT=1`
CMSISNN 需要开启这两个才能针对M4进行优化，不然他还是使用自己的纯C实现

xiongyu0523, 2019-09-29:
@majianjia, 感谢您提供的框架，我尝试在一个200MHZ的CM4F处理器上运行minist-simple。在没有开启CMSIS-NN的时，一次inference (运行nnom_predict的时间) 是900多ms。开启CMSIS-NN后是390ms。这两个数字正常吗？似乎没有达到ARM讲的4.x倍优化程度？ 

另外无论我在工程里是是否定义ARM_MATH_CM4, __FPU_PRESENT=1这两个宏（我已经验证他们是'真的'没有被定义) 速度都是390ms。我用的是CMSIS 5.6.0，里面好像已经没有用ARM_MATH_CM4。而FPU我已经在编译选项里写死了参数-mfloat-abi=hard -mfpu=fpv4-sp-d16，应该已经打开了

xiongyu0523, 2019-09-29:
上面速度慢的问题是因为没有开编译器优化，使用GCC -O3优化，不开CMSIS-NN为 100ms，打开CMSIS-NN为35ms。

Be997398715, 2019-09-30:
已经开启，所有环境都是搭配测试过的。这个原因可能是本身cmsis nn造成的吗？

Be997398715, 2019-09-30:
我一直觉得达不到4.x，在我的测试中，甚至只有1.x。。。

majianjia, 2019-09-30:
4x 是最高的时候。
不同的卷积参数会影响cmsis nn优化的能力。
具体参考一下cmsis nn的论文：
https://arxiv.org/pdf/1801.06601.pdf

Be997398715, 2019-09-30:
谢谢回复，我会仔细查阅的。顺带提一下，这个NNOM还是很好用的，我已经成功用来DIY自己的一个小玩具了

 https://arxiv.org/pdf/1801.06601.pdf