majianjia/nnom #72: kws编译成功了mcu端

tykjdxjcxq, 2019-12-13:
求解决方法
	layer[0] = Input(shape(63, 12, 1), nnom_input_data);
	layer[1] = model.hook(Conv2D(16, kernel(5, 5), stride(1, 1), PADDING_VALID, &conv2d_1_w, &conv2d_1_b), layer[0]);
	layer[2] = model.active(act_relu(), layer[1]);
	layer[3] = model.hook(MaxPool(kernel(2, 1), stride(2, 1), PADDING_VALID), layer[2]);
	layer[4] = model.hook(Conv2D(32, kernel(3, 3), stride(1, 1), PADDING_VALID, &conv2d_2_w, &conv2d_2_b), layer[3]);
	layer[5] = model.active(act_relu(), layer[4]);
	layer[6] = model.hook(MaxPool(kernel(2, 1), stride(2, 1), PADDING_VALID), layer[5]);
	layer[7] = model.hook(Conv2D(64, kernel(3, 3), stride(1, 1), PADDING_VALID, &conv2d_3_w, &conv2d_3_b), layer[6]);
	layer[8] = model.active(act_relu(), layer[7]);
	layer[9] = model.hook(Conv2D(32, kernel(3, 3), stride(1, 1), PADDING_VALID, &conv2d_4_w, &conv2d_4_b), layer[8]);
	layer[10] = model.active(act_relu(), layer[9]);
	layer[11] = model.hook(Dense(35, &dense_1_w, &dense_1_b), layer[10]);
	layer[12] = model.hook(Softmax(), layer[11]);
	layer[13] = model.hook(Output(shape(35,1,1), nnom_output_data), layer[12]);
打印出来的信息是这个应该咱们解决
NNoM version 0.3.0
Start compiling model...
Layer(#)         Activation    output shape    ops(MAC)   mem(in, out, buf)      mem blk lifetime
-------------------------------------------------------------------------------------------------
#1   Input      -          - (32840,1607,  72,)          (   756,     0,     0)    1 - - -  - - - - 
#2   Conv2D     - ReLU     - (32840,1607,  72,)          (     0,     0,  7200)    1 1 1 -  - - - - 
#3   MaxPool    -          - (32840,1607,  72,)          (     0,     0,     0)    1 1 1 -  - - - - 
-------------------------------------------------------------------------------------------------
WARNING: model returned at #0

learytomlin, 2020-01-12:
Try increasing the heap size. I had the same issue. Increasing heap solved it for me.

jonnor, 2024-03-30:
This looks to be answered and issue can be closed. CC @majianjia

closed 2024-04-07 by majianjia