majianjia/nnom #79: Use nnom project with LSTM Network

kouroshD, 2020-07-20:
Hi @majianjia . Thank you fir the project you have developed, It is very handy.
I am trying to use the code developed here with a LSTM network.
I saw in https://github.com/majianjia/nnom/blob/fc05b4aed3f06ff5eadb9f92c8ba33c0a94c5912/inc/layers/nnom_rnn.h#L38 you mentioned LSTM as well. So, I wanted to check if this has been implemented or not?
Also, I tried to convert the model I have learned to the header file as shown in :
https://github.com/majianjia/nnom/blob/fc05b4aed3f06ff5eadb9f92c8ba33c0a94c5912/examples/auto_test/main.py#L121-L127

Can you provide an example how to feed the input. I am giving the input to the  `generate_model` similar to the way I am giving to [`predict()`](https://www.tensorflow.org/api_docs/python/tf/keras/Model#predict) (or https://keras.io/api/models/model/)  function of but it is throwing errors. I saw, in the example for the time series data , at the end rnn models are not used.

Thank you.

majianjia, 2020-07-20:
hi, @kouroshD  thanks for your interest,
The RNN types (inc. GRU/LSTM)  are not currently supported because Keras does not support access to the intermediate output inside RNN cells. There is no straight forward method to quantize them. I am not sure I will/can implement it in the near future. 

If you definitely need to use RNN layers, you may build the RNN cells using the supported layers (Dense, Conv, Sigmoid, TanH..) then concatenate the outputs. In this way, the intermediate output can be explicitly quantized thus it should be able to convert using the scripts. 

Or, like many others just give up RNN types, and you can use [Temporal Convolutional Networks (TCN)](https://arxiv.org/pdf/1803.01271.pdf) (e.g. WaveNet) to deal with time sequence data (speech/sensor data). TCN layers are also good at dealing time sequence data well. It required regular Convolution with Dilation. It is already implemented in NNoM.

majianjia, 2020-08-25:
Hi @kouroshD 

I am not sure if you are still interested in RNN layers. I am now working on them 
Just make the simple RNN cell tested. It provides no impact on accuracy comparing to Keras' RNN, and is available on this [Develop branch](https://github.com/majianjia/nnom/tree/dev-0.4.x)
LSTM will be followed up in this week if everything goes well.

Thanks,
Jianjia

majianjia, 2020-08-29:
Hi @kouroshD 
LSTM and GRU are now working. Please check [uci-har-rnn example](https://github.com/majianjia/nnom/tree/dev-0.4.x/examples/uci-har-rnn)

kouroshD, 2020-09-12:
Hi @majianjia . Thank you for the updates and the support you have provided. In the following days, I will definitely look at them, and try to run them also my self.

Thanks,
Kourosh

jonnor, 2024-03-30:
This has been answered, and should be closed. CC @majianjia

closed 2024-06-01 by majianjia