majianjia/nnom #80: If I don't want use bias on Conv2D, What can i do?

passingduck, 2020-08-20:
I  have tried a lot of time for using bias on Conv2d.(take weights at ckpt  trained with  tensorflow_1.15)
(very dirty code....sorry)
[image]
[image]

First I changed nnom_utils.py. Erased all c_b line, and Changed layer.set_weights([c_w]).
And Changed related Length of convolution layer.

Twice I added zeros bias, But np.log2 functions make error.
So I added if statement like 'if min_value'==0: min_value=1e-36'

Both ways make this problem.
The First evaluate_model function works fine
[image]
But if I do the generate_model function and then do the evaluate function again, the evaluate_model function doesn't work fine.
[image]
And I'll show you a picture of the results on the board.
[image]
Thank you for reading the long article, and I apologize for terrible English and coding skills.
Safe Safety.

majianjia, 2020-08-20:
Hi passiduck,
Thanks for your interest. The simple answer is you have to use bias in both conv/dense layers.
This is restricted by the backend which is the CMSIS-NN. 
Any reason you dont want to use bias? I might add this feature later if it is necessary. 
Thanks,
Jianjia

passingduck, 2020-08-20:
Thanks for the quick reply.
I tried to use an existing trained model. Also Beacause beta in BatchNormalization works the same way, I read that no bias is needed.
If it is added in the future, I would appreciate it. Have a nice day.
Thanks!

passingduck, 2020-08-21:
Hi jianjia.
I used nnom after I added bias,But I had same issues.
Can you infer any possible reason for this problem to occur?
Before generate_model image 
[image]
After generate_model image
[image]

After using generate_model function, the first layer's weights changes like this. Could this be a problem?
Before
[image]
After
[image]

Thanks, Have a great weekend!
passingduck

piyush-das, 2021-12-28:
Hi @majianjia For support of legacy models, would it be possible to add support for having a no bias in the Conv Layer, or alternatively if their is a hacky way to implement the same, could you please give pointers.

Thanks.

majianjia, 2021-12-28:
nnom doesn't support bias because the backends must have one to work properly currently. 

You must add it somewhere. There are 2 ways to do it. 
1. modify nnom conv layers, when there isn't bias available, malloc a chunk of memory at the size of output, set it to 0. use it as bias.  
2. In python code, after loading model (the model after training.). Add a bias tensor to the layer that doesn't have bias. Set it to the correct shape with all values zero. 

Hope it helps

piyush-das, 2021-12-28:
Thanks for the pointers.
I tried option 2. However, it resulted in
```log
File "/home/xyz/nnom/scripts/nnom.py", line 216, in find_dec_bits_max_min
  int_bits = int(np.ceil(np.log2(max(max_val, min_val))))
OverflowError: cannot convert float infinity to integer
```
Then just for cleaning the flow , I also introduced a bias_initializer as "random_normal", so that the max_val, min_val don't reduce to 0 for the bias layers. However, then I got the following error : 
```log
File "/home/xyz/nnom/scripts/nnom.py", line 731, in quantize_weights
    f.write('#define ' + layer.name.upper() + '_BIAS_LSHIFT '+to_cstyle(bias_shift) +'\n\n')
UnboundLocalError: local variable 'bias_shift' referenced before assignment
```

piyush-das, 2021-12-28:
It appears that nnom uses the name of the layers instead of the instance type. In the case that the layers have a different nomenclature, then the bias_shift variable doesn't get assigned.