majianjia/nnom #83: 0 to 1 output (binary_crossentropy)

sifourquier, 2020-09-03:
Hello my model have 1 output with a value from 0.0 to 1.0

```
x=Dense(1)(x)
predictions=Activation('sigmoid')(x)
```

then when my float model ouput are 1 my optimized model output are 1 but when my float model ouput are 0.99 my optimized model output are 0.

I do somthing wrong or is a bug? how to have a outpour form 0 to 255 or 0 to 127?

If you can help my thanks you.

majianjia, 2020-09-03:
If you are using the current Master branch, there is a bug in getting `prob`. The latest `0.4.x-dev` branch should have correct this. 
The below code is corrected version.
https://github.com/majianjia/nnom/blob/a039be030738ec6e9202c46cb9bde32d844db1af/src/core/nnom_utils.c#L297

If you want to stay with the current master branch: 
After running the model, you may check the raw output to see the value which is in `nnom_output_data[0]`.

Since you are using the sigmoid as the final layer, the value will be q0.7 format which means that decimal bit is 7. 
So we are expecting the range from 0~127 to represent 0-1.

sifourquier, 2020-09-03:
Thanks you but i check the bug and is only for multiple output and i have try with dev-0.4.x same result
But i have found some value with float model ouput = >0.5 <1 and optimized model = 1

Then i thinks is a other probleme (nnom_output_data[0] return 0 or 127 never other value)

Maybe is only my model loos many accurate with int8 optimization.

majianjia, 2020-09-04:
The 0/127 results might caused by the sigmoid was forced to saturate the output because the num of the integral bit of the dense's output data is > 3, e.g. if int = 4, then the dec bit is 7-4=3. In the cmsis-nn's (also the local backend) tanh and sigmoid, they don't support this number so we have to saturate the output to 1 and 0 (for sigmoid).

You can use `model_io_info()` to list the Q format and see if the output of the dense int bit is > 3. 

If it dose, there is not much we can do now to get better resolution, unless 16bit or float but they are not yet implemented.
340 to one might cause the value too large. you can try to add another dense with smaller num between. 
Or try not to use activation. just use the raw dense output.