majianjia/nnom #87: The inputs and outputs for the XOR problem.

miaowrx, 2020-09-12:
Hi @majianjia 👍

I try to build a simple neural network to predict the outputs of XOR logic like this:

[image]

And generate the "weights.h" use the following code:

[image]

finally, run the neural network:

```c
#include "weights.h"

nnom_model_t* model;

void main(void)
{
        model = nnom_model_create();

        nnom_input_data[0] = 0;
        nnom_input_data[1] = 0;
        model_run(model);

        nnom_input_data[0] = 0;
        nnom_input_data[1] = 1;
        model_run(model);

        nnom_input_data[0] = 1;
        nnom_input_data[1] = 0;
        model_run(model);

        nnom_input_data[0] = 1;
        nnom_input_data[1] = 1;
        model_run(model);
}
```
The inputs are [0, 1] and the outputs should be [1] but it's [106]. so where I am wrong？

[image]

majianjia, 2020-09-12:
The input of your Keras model is [0-1] therefore you can see the quantise decimal bit is [7]. Also, your output is sigmoid also the same decimal bit =7.
[image]

That decimal bits means, how may bit in the int8_t is used for defining the decimal part of the number. Or on other work, if the fixed-point num is 106, and its decimal bit n=7, the corresponding floating-point number is [106 / (2^n-1)] = 0.828.

Same as your input, you need to convert [0-1] to [0-127]. You should put 127 into nnom_input_data[] instead of putting 1 directly.

miaowrx, 2020-09-13:
Thanks, I use the new code below:

```c
#include "weights.h"

nnom_model_t* model;

void main(void)
{
    model = nnom_model_create();

    nnom_input_data[0] = 0;
    nnom_input_data[1] = 0;
    model_run(model);

    printf("%03d\t%03d = %d\r\n", nnom_input_data[0], nnom_input_data[1], nnom_output_data[0]);

    nnom_input_data[0] = 0;
    nnom_input_data[1] = 127;
    model_run(model);

    printf("%03d\t%03d = %d\r\n", nnom_input_data[0], nnom_input_data[1], nnom_output_data[0]);

    nnom_input_data[0] = 127;
    nnom_input_data[1] = 0;
    model_run(model);

    printf("%03d\t%03d = %d\r\n", nnom_input_data[0], nnom_input_data[1], nnom_output_data[0]);

    nnom_input_data[0] = 127;
    nnom_input_data[1] = 127;
    model_run(model);

    printf("%03d\t%03d = %d\r\n", nnom_input_data[0], nnom_input_data[1], nnom_output_data[0]);

}
```

But the output is still strange:

```
Model version: 0.4.1
NNoM version 0.4.1
Data format: Channel last (HWC)
Start compiling model...
Layer(#)         Activation    output shape    ops(MAC)   mem(in, out, buf)      mem blk lifetime
-------------------------------------------------------------------------------------------------
#1   Input      -          - (   2,  85,   7,)          (  1190,  1190,     0)    1 - - -  - - - -
#2   Dense      - Sigmoid  - (   5,          )     5950 (  1190,     5,  2380)    1 1 1 -  - - - -
#3   Dense      - Sigmoid  - (   5,          )       25 (     5,     5,    10)    1 1 1 -  - - - -
#4   Dense      - Sigmoid  - (   1,          )        5 (     5,     1,    10)    1 1 1 -  - - - -
#5   Output     -          - (   1,          )          (     1,     1,     0)    - - 1 -  - - - -
-------------------------------------------------------------------------------------------------
Memory cost by each block:
 blk_0:1192  blk_1:2380  blk_2:8  blk_3:0  blk_4:0  blk_5:0  blk_6:0  blk_7:0
 Total memory cost by network buffers: 3580 bytes
000     000 = 126
000     127 = 126
127     000 = 125
127     127 = 125
```

The strangest thing is that the output is different every time I run this program:

```
Model version: 0.4.1
NNoM version 0.4.1
Data format: Channel last (HWC)
Start compiling model...
Layer(#)         Activation    output shape    ops(MAC)   mem(in, out, buf)      mem blk lifetime
-------------------------------------------------------------------------------------------------
#1   Input      -          - (   2,  85,   7,)          (  1190,  1190,     0)    1 - - -  - - - -
#2   Dense      - Sigmoid  - (   5,          )     5950 (  1190,     5,  2380)    1 1 1 -  - - - -
#3   Dense      - Sigmoid  - (   5,          )       25 (     5,     5,    10)    1 1 1 -  - - - -
#4   Dense      - Sigmoid  - (   1,          )        5 (     5,     1,    10)    1 1 1 -  - - - -
#5   Output     -          - (   1,          )          (     1,     1,     0)    - - 1 -  - - - -
-------------------------------------------------------------------------------------------------
Memory cost by each block:
 blk_0:1192  blk_1:2380  blk_2:8  blk_3:0  blk_4:0  blk_5:0  blk_6:0  blk_7:0
 Total memory cost by network buffers: 3580 bytes
000     000 = 125
000     127 = 124
127     000 = 123
127     127 = 123
```

```
Model version: 0.4.1
NNoM version 0.4.1
Data format: Channel last (HWC)
Start compiling model...
Layer(#)         Activation    output shape    ops(MAC)   mem(in, out, buf)      mem blk lifetime
-------------------------------------------------------------------------------------------------
#1   Input      -          - (   2,  85,   7,)          (  1190,  1190,     0)    1 - - -  - - - -
#2   Dense      - Sigmoid  - (   5,          )     5950 (  1190,     5,  2380)    1 1 1 -  - - - -
#3   Dense      - Sigmoid  - (   5,          )       25 (     5,     5,    10)    1 1 1 -  - - - -
#4   Dense      - Sigmoid  - (   1,          )        5 (     5,     1,    10)    1 1 1 -  - - - -
#5   Output     -          - (   1,          )          (     1,     1,     0)    - - 1 -  - - - -
-------------------------------------------------------------------------------------------------
Memory cost by each block:
 blk_0:1192  blk_1:2380  blk_2:8  blk_3:0  blk_4:0  blk_5:0  blk_6:0  blk_7:0
 Total memory cost by network buffers: 3580 bytes
000     000 = 125
000     127 = 126
127     000 = 125
127     127 = 125
```

majianjia, 2020-09-13:
The input shape doesn't look correct please check

miaowrx, 2020-09-14:
I set input shape is (1,2) and everything is right, thanks.

closed 2020-09-14 by miaowrx

majianjia, 2020-09-14:
Good！