majianjia/nnom #88: 1D vector problem

giusseppeone, 2020-09-16:
Hi Majianjia ,
I tested NNOM and ran into such a problem. I created a primitive model for the implementation of the XOR problem. 
[image]
After train and generate model the values appear to be OK
[image]
[image]
After test on STM32G431 I found this problem (input is one dim vector as created in jupyter => input shape(2,) )
[image]
and output from console is
[image]
I tried to modify the code to test the functionality
[image]
after run the code output is OK ane value is ok too. 
[image]
My question is , can work NNOM with input shape=(2,) ?
Thank you , nnom is excellent job :)

majianjia, 2020-09-16:
The current Input layer only can take 3d shape data. For 2D data, the Input will still expend the shape to 3d.
But there is no implementation to expend 1D data to 3D, which is also not a good solution for the future. So I didnt implement this. 
This is an very old idea which NNoM was primary only focus on 2D image (3D data)

I recently look back to this issue and are planning to change it.
In the future version I will make Input to support native 1D and 2D data. But it requires some works to do.

The current solution for 1D data is to at least expend the input shape in keras to 2d then use flatten() in keras to convert it to 1D. 
Just use input_shape=(1, x) then use a flatten following by it.

giusseppeone, 2020-09-16:
Thanks for answer , 
I tested with input_shape =(1,x) it works well , 
many thanks

closed 2020-09-16 by giusseppeone