majianjia/nnom #89: Loss of accuracy

BaptisteNguyen, 2020-09-16:
For a project, i have trained 2 CNN (Lenet and resnet) on the dataset GTSRB. Then I used Nnom to put them on microcontrollers.
The accuracy of Lenet is 99,4% on tensorflow and 99,1% on microcontroller.
But the accuracy of resnet is 99.3% on tensorflow and around 50% on microcontroller.
Can you look into my project (https://github.com/BaptisteNguyen/testNNom) in order to explain this difference?

majianjia, 2020-09-16:
I briefly checked your Restnet model. There are some trick you might want to try

Increase your first layer's channel might improve the performance. 

In `nnom_port.h` turn on/off the `NNOM_TRUNCATE ` option to see if there is any impovment. This should be aligned with CMSIS-NN. There is a known issue i am trying to fix which this option sometimes it cannot swith to the correct one. 

If you can use the new script `nnom.py` instead of  `nnom_utils.py` to generate the model. i.e. use `from nnom import *`
In the new script,  you can turn on per_channel_quantoption to see if the result is better. The other option you can try is set `quantize_method='kld'` which some time will give bettery accuracy in deep network. 
`generate_model(model, x_test, per_channel_quant=False, name='weights.h', format='hwc', quantize_method='max_min')`

Hope it help,
Please let me know how it is going, thanks

BaptisteNguyen, 2020-09-18:
My original resnet has an accuracy of 43.6% on microcontroller (tested on 500 images).
I have tested the trick you proposed.
I have doubled the number of channel in the neural network. It reaches an accuracy of 53.7%.
Turning on/off NN_TRUNCATE flags doesn't change the accuracy.
With per channel quantization, the model has an accuracy of 44%.
With kld quantization, the model has an accuracy of 33.9%.
With kld and per channel quantization, the model has an accuracy of 43.6%.
I have also tried a version of the resnet without batch-normalization but it gives me an accuracy of 37.4%.

During my test, I have a few issue:

1) In nnom.py, in the function quantize_data, I have replace d = np.round(d * 2 ** dec_bits[i) by d = np.round((d * 2 ** dec_bits[i]).astype(float)). When d values are all close to 0, dec_bits[i] can have very high value (more than 100 in some cases). When  dec_bits[i] is high, d * 2 ** dec_bits[i] returns a numpy array of object instead of float. Np.round doesn’t support numpy array of object which leads to an error.

2) generate_model in nnom.py doesn’t work with model which are loaded or copied. Because of this I need to retrain my network for each test.
The problem seems to be in lines suchas :
	- inp = layer_name_from_tensor(layer.input)  LI[inp]
	- iname = layer.input[0].name.split('/')[0] dec_min = layer_q_list[iname][0]
The keys of dictionary like LI or layer_q_list are the name of the different layer(layer.name). For a model not loaded or copied, the line of code « layer.input[0].name.split('/')[0] » gives the name with of the input layer. But for a loaded or copied model, this line of code doesn’t give the name of the input layer. I have tried to replace « layer.input[0].name.split('/')[0] » by « layer.inbound_nodes[0].inbound_layers.name ».The code run but they are some difference between the file .h generated.
I have add a file nnom2.py in the github repository https://github.com/BaptisteNguyen/testNNom.

Baptiste Nguyen

majianjia, 2020-09-19:
Thanks @BaptisteNguyen 
Sorry I am not in home this weekend so I couldnt debug the code.
The current script doesnt work with copied model so I always save, delete the model instance , then reload it before conversion. Tthe current script is pretty much rely on the name of the layer,  and the Keras layer naming is inconsistence, which causes this copied model issue.

For the accuracy problem, it might due to the quantization limitation. Merging layers(concat, add, sub, mult) are requiring each input must have same q format (Same decimal bit width). sometime it Will force some of the layers to output on a lower resolution(lower decimal bit. it will sometime lead to huge accuracy drop. 

Or there might be some bugs on the conversion. thanks for providing the code and model. I will use them to debug once i am back

Thanks,
Jianjia

majianjia, 2020-10-08:
Hi @BaptisteNguyen ,
I have checked and i cant find where is the issue. 
But what I can see is due to the `add` of ResNet which is not working well in 8bit width. Because the input of both data in `add` (same as concat/sub/mult) are forced to have the same Qformat. This might introduce many loss. 

You might reduce the training epoch to a few and see if there is any different. Sometime training for too many epoch will lead to extreme values (activations) that affect the quantisation range.

BaptisteNguyen, 2020-10-14:
Thank you for your answer.
I have put this project on standby, but i will try to find the issue in the future.