majianjia/nnom #94: rnn-denoise model.predict

sc-learner, 2020-10-13:
I met with a Error when running rnn-denoise example,  "ValueError: Input 0 is incompatible with layer gru: expected shape=(2048, None, 40), found shape=[32, 1, 40]"
This Error occurs in the function voice_denoise: prediction = model.predict(feat)
And it also happens when generating weight file (model.predict also used there).

After I checked tensorflow/python/keras/engine/training.py, I modified main.py from "prediction = model.predict(feat)" to "prediction = model.predict(feat,batch_size=timestamp_size)". It seemed OK now. 

So, is this modification OK or not? And if it's OK, then how to modify nnom.py?

majianjia, 2020-10-13:
For the Keras, the inference batch size (32) should be the same as the training batch size (2048). 
NNoM dosent require this because it take 1 sample each time. (batchsize=1)

2048 is equvalant to around 32 seconds, which might be too long for some audio.

sc-learner, 2020-10-13:
Thank you! But How can I avoid this Error if I don't modify the code?

majianjia, 2020-10-13:
What you did is correct, I will check later When I am back.

sc-learner, 2020-10-14:
> Hi @sc-learner
> 
> It's weir the code dump this error because the code is already cutting the signal if the lenght is cannot be divided exactly by the size of timestamp.
> 
> It looks like you are processing an audio signal that doesn't long enough (like the training data)
> You may need to pad the signal with 0 and extend it to at least 32 seconds when passing it to voice.

So you mean you haven't met this Error? I didn't modify any code from github (except a little directory part) when I met the Error. And I printed the shape of the data I fed to model.predict, it said "(2048,1,40)", exactly the same as the model required. But then the Error pointed out the data I fed was "(32,1,40)" until I added that "batch_size=timestamp_size"...

By the way, my tf version is 2.3.0 and 2.3.1. I wonder if that matters?

majianjia, 2020-10-14:
I am using tf2.1 and tf1.15 Which didnt throw this error. might be some changes over the default argument of predict()

and yes, please use batch_size=timestamp at the moment. I will add this for the next commit

sc-learner, 2020-10-15:
> I am using tf2.1 and tf1.15 Which didnt throw this error. might be some changes over the default argument of predict()
> 
> and yes, please use batch_size=timestamp at the moment. I will add this for the next commit

Thank you!

FYI.
I also tried tf1.15, but I got the following Error when doing generate_model function (tf 2.3 is OK here):

Traceback (most recent call last):
  File "main.py", line 333, in <module>
    main()
  File "main.py", line 322, in main
    generate_model(model, x_train[:timestamp_size*4], name='weights.h')
...
tensorflow.python.framework.errors_impl.InvalidArgumentError: input_1:0 is both fed and fetched.

majianjia, 2020-10-16:
Looks like the support for tf1 is still not that stable.

kouohhashi, 2020-11-04:
@sc-learner 
I have a same issue.
How didi you fix the generate_model error?
Could you share the fix to save your model?

Thanks in advance.

sc-learner, 2020-11-04:
> @sc-learner
> I have a same issue.
> How didi you fix the generate_model error?
> Could you share the fix to save your model?
> 
> Thanks in advance.

can't remember clearly now.

tf 2.3.1

in main.py:
def voice_denoise
prediction = model.predict(feat) --> prediction = model.predict(feat,batch_size=timestamp_size)

in nnom.py:
def quantize_output
features = layer_model.predict(x_test) --> features = layer_model.predict(x_test,batch_size=2048)

kouohhashi, 2020-11-04:
@sc-learner 
Thanks. it worked!