majianjia/nnom #99: Problem running examples (bad results / python errors)

Mjonir, 2020-11-05:
Hello, 

I have been trying to run the included examples but have ran into issues:

**mnist_simple.py** ran to completion after I applied the cudNN fix and renamed 'acc' and 'val_acc' to 'accuracy' and 'val_accuracy' respectively. I attach the (anonymized) log and self-evaluation results which are as expected. However, when I try to run the generated quantized network in C, the results are heavily skewed towards one label. I attach the run log, and a simple C file with Makefile to reproduce the issue:
[mnist-simple.zip](https://github.com/majianjia/nnom/files/5495928/mnist-simple.zip)

**auto_test.py** crashes during execution, here is the execution log:
[auto_test_log.txt](https://github.com/majianjia/nnom/files/5495853/auto_test_log.txt)

Thank you in advance

majianjia, 2020-11-05:
Hi @Mjonir , 
May I know the TF version you are using? recommened TF2.2+. For the crash, are you setting Keras to 'CHW' or 'channel first'? The script currently only support the default `HWC` shape format. 

I am checking both attachement you provided. 

Would you use the example [weights.h](https://github.com/majianjia/nnom/blob/master/examples/mnist-simple/weights.h) in the example folder other that the one you converted from you model to test? see if the prediction is correct?  for the mnist-simple example. 

Thanks,
Jianjia

Mjonir, 2020-11-06:
Hello,
I am using tensorflow 2.3.1, but I was indeed setting Keras to channels_first in keras.json from previous experiments with CMSIS-NN (forgot about that one!). After fixing this (and applying the cudNN fix), both scripts ran with the expected results.

For information, with multiple GPUs I have to use this cudNN fix (loop over GPUs) or I get the error below:
```
for device in physical_devices:
        tf.config.experimental.set_memory_growth(device, True)
```
`ValueError: Memory growth cannot differ between GPU devices`

Also for informaiton, the included weights.h failed at compile-time, from what I assume is an inconsistency with the recent interface changes. Here is the full compile log:
[compile_log.txt](https://github.com/majianjia/nnom/files/5501120/compile_log.txt)

Thank you very much for the support, I will be moving on to trying it on my own model

closed 2020-11-06 by Mjonir