majianjia/nnom #111: rnn-denoise did not achieve the desired results

poweel, 2021-01-23:
I have test the demo on ubuntu16.04, but I got a bad result, some voice was cutoff and the noise of keyboard was not removed. while test with rnnoise, it got a good result.
I just checkout the code and run scons to make the demo, the input file was 16k 16bit mono noisy voice file. 
The question is :
1. the training was must or I can use the model in git source code?
2. in article "https://zhuanlan.zhihu.com/p/275644326", says there are 4 files, where can I get the 4 files so I can test the demo file and compare the result?
    _noisy_sample.wav  --> 原始带噪音文件
    _filtered_sample.wav  --> 用真实gains降噪的文件(训练的gains)
    _nn_filtered_sample.wav   --> Keras浮点模型gains 降噪
    _nn_fixedpoit_filtered_sample.wav   --> NNoM定点模型gains降噪
3. Are there some parameter to changed to improve the result?

poweel, 2021-01-23:
The log info:
./rnn_denoise fip11c_16k.wav fip11c_out_16k.wav 
Model version: 0.4.3
NNoM version 0.4.3
To disable logs, please void the marco 'NNOM_LOG(...)' in 'nnom_port.h'.
Data format: Channel last (HWC)
Start compiling model...
Layer(#)         Activation    output shape    ops(MAC)   mem(in, out, buf)      mem blk lifetime
-------------------------------------------------------------------------------------------------
#1   Input      -          - (   1,   1,  40,)          (    40,    40,     0)    1 - - -  - - - - 
#2   RNN/GRU    -          - (   1,  96,     )      86k (    40,    96,  1808)    1 1 1 -  - - - - 
#3   RNN/GRU    -          - (   1,  96,     )     102k (    96,    96,  1920)    1 1 1 -  - - - - 
#4   RNN/GRU    -          - (   1,  48,     )      32k (    96,    48,  1056)    1 1 1 -  - - - - 
#5   Flatten    -          - (  48,          )          (    48,    48,     0)    - - 1 -  - - - - 
#6   Dense      - HrdSigd  - (  20,          )      960 (    48,    20,    96)    1 1 1 -  - - - - 
#7   Output     -          - (  20,          )          (    20,    20,     0)    - 1 - -  - - - - 
-------------------------------------------------------------------------------------------------
Memory cost by each block:
 blk_0:1920  blk_1:1808  blk_2:96  blk_3:0  blk_4:0  blk_5:0  blk_6:0  blk_7:0  
 Memory cost by network buffers: 3824 bytes
 Total memory occupied: 7512 bytes

Processing file: fip11c_16k.wav

Print layer input/output..
Layer(#)        -  Input(Qnm)  Output(Qnm)   Oshape 
----------------------------------------------------------
#1  Input      -    0. 7      0. 7      (   1,   1,  40,)
#2  RNN/GRU    -    0. 7      0. 7      (   1,  96,     )
#3  RNN/GRU    -    0. 7      0. 7      (   1,  96,     )
#4  RNN/GRU    -    0. 7      0. 7      (   1,  48,     )
#5  Flatten    -    0. 7      0. 7      (  48,          )
#6  Dense      -    0. 7      0. 7      (  20,          )
#7  Output     -    0. 7      0. 7      (  20,          )

Print running stat..
Layer(#)        -   Time(us)     ops(MACs)   ops/us 
--------------------------------------------------------
#1  Input      -         0                  
#2  RNN/GRU    -         0          86k     
#3  RNN/GRU    -         0         102k     
#4  RNN/GRU    -         0          32k     
#5  Flatten    -         0                  
#6  Dense      -         0          960     
#7  Output     -         0                  

Summary:
Total ops (MAC): 222480(0.22M)
Prediction time :0us
Total memory:7512

Noisy signal 'fip11c_16k.wav' has been de-noised by NNoM.
The output is saved to 'fip11c_out_16k.wav'.

poweel, 2021-01-23:
when I run the prebuild demo, there is an error:
./rnndenoise fip11c_16k.wav fip11c_16k_out.wav 
Model version: 0.4.3
Please set static memory using 'nnom_set_static_buf()' before calling model created.

poweel, 2021-01-28:
I have fixed it, because all the training voice is about -6db, but the voice in our device maybe about -12db, -18db, current model weights can not cover this case, I retraining the model with -18db,0db random scale, and it worked.

closed 2021-01-28 by poweel

wrz1999, 2024-01-16:
> when I run the prebuild demo, there is an error: ./rnndenoise fip11c_16k.wav fip11c_16k_out.wav Model version: 0.4.3 Please set static memory using 'nnom_set_static_buf()' before calling model created.

How to fix it? Can you tell me？ Thank you very much