majianjia/nnom #112: rnn-denoise dump denoise_weights.h files failed after training

jagger2048, 2021-01-31:
Hi，
  I try to train the rnn-denoise demo, but it failed when generating the NNoM model in quantize_output function. Is there any update for this demo?
 Here is the log
...
filtering with frequency:  5394.2623288811965
filtering with frequency:  6162.04951128775
filtering with frequency:  7026.566556252747
/root/miniconda3/envs/myconda/lib/python3.5/site-packages/matplotlib/figure.py:448: UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
  % get_backend())
input_1 Quantized method: max-min  Values max: 1.0 min: -1.0 dec bit 7
================
[<tensorflow.python.keras.engine.input_layer.InputLayer object at 0x7fd630553ac8>]
1
Traceback (most recent call last):
  File "main.py", line 315, in <module>
    main()
  File "main.py", line 297, in main
    generate_model(model, x_train[:timestamp_size*4], name='denoise_weights.h')
  File "/mnt/nnom/scripts/nnom.py", line 750, in generate_model
    layer_q_list = quantize_output(model, x_test, layer_offset=False, quantize_method=quantize_method)
  File "/mnt/nnom/scripts/nnom.py", line 498, in quantize_output
    layer_model = Model(inputs=model.input, outputs=in_layer.output)
AttributeError: 'list' object has no attribute 'output'
(myconda) root@f8209e3ef1a3:/mnt/nnom/examples/rnn-denoise# 

B&R

majianjia, 2021-01-31:
hi @jagger2048 
The script you are using seems different than the one in master branch. 
What is the nnom version you are using? and what is your tf version. The current script only tested with TF2.1+
Thanks

jagger2048, 2021-02-01:
Hi,
  I use the dev-0.4x branch and TF1.3+ in my test...:) I will try the master branch and the expected TF version.
B&R

closed 2021-02-01 by jagger2048