majianjia/nnom #114: Crash in tensor_size() on ESP32

junfengzh, 2021-02-15:
I managed to get the keyword_spotting example running on ESP32 (ESP32-PICO-D4 of M5StickC), as a MicroPython module. It works great, except that sometimes it crashes in tensor_size(). Here is the call stack:

```
0x401f0612: tensor_size at ??:?
0x40118b29: input_run at ??:?
0x40118a69: model_run_to at ??:?
0x40118a89: model_run at ??:?
0x40117cc4: nnom_predict at ??:?
```

Note that the file names and line numbers are not available, as the sources are compiled using `-Ofast` to maximize speed optimization.

Seems there's a bug somewhere in the code. Do we have idea what that could be?

majianjia, 2021-02-15:
The tensor_size() is multiplying the N shape data, with length=N. 
The bug might be a Null tensor passed in. Could you try to reduce the optimization level to see if it helps?

junfengzh, 2021-02-16:
Thanks for your reply. When I reduce the optimization level, the process runs much slower (500ms+ for a 1-second audio, vs. 200ms optimized), and problem happens elsewhere in MicroPython. If I replay pre-recorded audio, it works just fine. Basically I could not get a stable repro of the issue.

Instead of crashing, can you fix the code to fail silently when the input is invalid? E.g. nnom_predict returns no label and 0 probability? Ideally you can return an error code.

majianjia, 2021-02-17:
@junfengzh 
Good suggestion, I haven't touched these application APIs for so long. 
Besides, I recommend to retrieve the output directly from the buffer and do your own calculation of the probability, since the current API provides only basic calculation.

junfengzh, 2021-02-17:
Yes I actually want the top N labels instead of one. While I'm reading the code, if this is a small fix for you, it would be great if you can make it.