majianjia/nnom #115: Different output prediction with CMSIS-NN as backend

angelocatalani, 2021-02-18:
Hi,
thanks for your great work!.

Do you have any ideas on why the prediction is always wrong when the `NNOM_USING_CMSIS_NN` is defined ?.

Without that optimization, the predictions are correct:
While in that case I always get the output predictions (4 classes with Softmax) always the same value: `32`.
 
 My model is a ResNet with 1D CNN and 1D max pool.
 This is how I used the API: `generate_model(model, x_test, name='quant_model.h', format='hwc')`.

majianjia, 2021-02-18:
Hi @angelocatalani 
Could you explain more about the issue?
Is “Without that optimization” means the compiler's optimization option?
Thanks

angelocatalani, 2021-02-18:
"Without that optimization" means the macro `NOM_USING_CMSIS_NN`  inside `port.h` is not defined:

```c
/* Backend selection */
//#define NNOM_USING_CMSIS_NN       // uncomment if use CMSIS-NN for optimation 
```

majianjia, 2021-02-20:
which CMSIS version are you using? I am not sure if there is some changes since my last check. 
The ones I used were CMSIS 5.6.0 and 5.7.0

angelocatalani, 2021-02-20:
thank you, I was using the `5.7.0` version. Downgrading to the `5.6.0` version seems to have solved the issue

closed 2021-02-20 by angelocatalani

majianjia, 2021-02-20:
Thanks for testing, I will try to find the issue related to the cmsis 5.7.0