majianjia/nnom #116: 'BatchDataset' object has no attribute 'max'

Technerder, 2021-03-04:
Running into the following error while trying to use `nnom.generate_model(model, test_data)` 

```
Traceback (most recent call last):
  File "C:/Users/Technerder/Dev/TimeSeriesClassification/train_lstm_1d.py", line 60, in <module>
    nnom.generate_model(model, test_data)
  File "C:\Users\Technerder\Dev\TimeSeriesClassification\nnom.py", line 746, in generate_model
    layer_q_list = quantize_output(model, x_test, layer_offset=False, quantize_method=quantize_method)
  File "C:\Users\Technerder\Dev\TimeSeriesClassification\nnom.py", line 527, in quantize_output
    dec_bits = find_dec_bits_max_min(features, bit_width=8)
  File "C:\Users\Technerder\Dev\TimeSeriesClassification\nnom.py", line 214, in find_dec_bits_max_min
    max_val = abs(data.max()) - abs(data.max()/pow(2, bit_width)) # allow very small saturation.
AttributeError: 'BatchDataset' object has no attribute 'max'
```

Does nnom not support the usage of batched datasets?

majianjia, 2021-03-04:
Hi @Technerder  
I don't think the current script support batch dataset.

Technerder, 2021-03-04:
Dang, are there any plans to add support for them in the near future?

majianjia, 2021-03-05:
I havent use batchdataset before. Could you use a regular dataset for the calibration?