majianjia/nnom #123: 在Stm32上运行minist例程发现的错误

NeroAsmarr, 2021-04-27:
我的设备为stm32h743vit6，核心板来自：https://gitee.com/WeAct-TC/MiniSTM32H7xx。
环境为cubemx生成hal库，然后导入mdk进行编译的。

在运行minist例程的时候，使用了如下的验证方式：
[image]

然后运行报错告诉我网络结构中的Conv2d参数少了一个，我查看源代码后发现相较于教程中多了一个dilation参数，如图：
[image]

然后我查看了添加了keras文档后其默认的dilation参数为(1, 1)，所以我直接复制了前面的stride(1, 1)作为其参数（虽然我不知道是否正确），如图：
[image]

之后便可以不报错烧录进单片机，但是我通过串口结果发现运行到Input层到p5点的时候就停止运行了：
[image]

p5点我设置在compile_layer函数的输出dim的地方：
[image]

我看代码里dilation也是nnom_3d_shape_t类的，不知道哪里出问题了。

NeroAsmarr, 2021-04-27:
我又测试了其他几个ministdense和uci-inception等，都是有这个问题，卡在同样的位置

majianjia, 2021-04-27:
脚本和nnom版本需要使用同一个。
你的weights好像是0.2的脚本转的，需要换成最新的

NeroAsmarr, 2021-04-28:
我尝试了最新的4.3脚本训练与生成mnist模型，串口结果是匹配的，同样是停在了相同位置。
[image]

NeroAsmarr, 2021-04-28:
> 脚本和nnom版本需要使用同一个。
> 你的weights好像是0.2的脚本转的，需要换成最新的

我尝试了最新的4.3脚本训练与生成mnist模型，串口结果是匹配的，同样是停在了相同位置。
[image]

majianjia, 2021-04-28:
可能是stack 空间不足。可以尽量改大试试。

或者使用静态内存，参考这里。
https://github.com/majianjia/nnom/blob/5a5dee5effe02069e09854a9a6f5a2f3689d9db2/examples/auto_test/main.c#L69
并且打开port.h里面的静态内存选项

NeroAsmarr, 2021-04-28:
谢谢🙏，我配置静态内存后成功运行了。

但是我还是想了解一下配置stack，然后使用cubemx的配置选项配置minimum stack size，但是一直把握不住度，要么溢出要么还是不够，请问有配置的一些方法或者相关链接可以参考的吗？

majianjia, 2021-04-28:
我这边一直都是用RT Thread来调试，默认会把所有的剩余内存当成stack使用。
默认使用libc的内存接口。我不太清楚cubemx的malloc实现。这个需要很大才可以，跟模型本身层大小有关。

NeroAsmarr, 2021-04-28:
了解！谢谢

closed 2021-04-28 by NeroAsmarr