majianjia/nnom #130: Hard Fault error

Antimoon51, 2021-05-06:
Hi,

I am trying to build an application using NNoM 0.4.2 and CMSIS 5.7.0 on the [STM32F723E-Disco](https://www.st.com/en/evaluation-tools/32f723ediscovery.html) Board. I use the Keil µVision IDE V5.33.

I am encountering a hard fault error during debugging. As far as I can tell the hard fault is called from the `conv2d_build` funktion in the file nnom_conv2d.c.

I have already tried using different networks in an empty project (despite CMSIS and NNoM).

majianjia, 2021-05-07:
Hi @Antimoon51 
Might be there is not enough stack. You will need to set it to as much as possible. 
Or change it to static buffer and again set it to large size first then reduce it according to the log printed out.

Antimoon51, 2021-05-20:
I think I worked this out. It wasn't as much about the stack-, but about the heap size. In my case I could leave the stack size at the initial value, but increased the heap size by a lot and in the end by what the log printed after compiling the network. I'm going to close this issue as the initial problem has been solved.

closed 2021-05-20 by Antimoon51

majianjia, 2021-05-20:
@Antimoon51 
Sorry, it was my fault. I was confused between heap and stack....