majianjia/nnom #137: 'KerasTensor' object has no attribute 'input'

alimynn, 2021-07-17:
I have a dense network with a InputLayer as the first layer followed by Dense layers and ReLU layers. At the end, there is a Activations layer with sigmoid activation function as can be seen in my screenshot.

[image]

I have tensorflow version 2.5.0 installed and when executing generate_model, i get the error "'KerasTensor' object has no attribute 'input'" in file nnom.py in line 827. Is this an error with the tensorflow version, a bug in the library or did i define anything wrong?

Thanks

P.S.: This is the stack trace and the model definition.
[image]

[image]

majianjia, 2021-07-17:
Hi @alimynn 
Please try to use `layers.Input` instead of `layers.InputLayer` and see if it helps. 
Other seems fine from me. 
If it doesn't help, please could you let me know your TF version? I recommend using the version above 2.1. I normally test with 2.2 and 2.3
Thanks,

alimynn, 2021-07-17:
Hi and thank you for your fast answer. I have TF version 2.5 installed first and tested now also with version 2.2. With 2.2 the error comes in line 825 as you can see in the screenshot below.

[image]

Also, changing from layers.InputLayer to layers.Input makes no difference.
What i found now is that when i change the model definition like in the next screenshot, everything works fine.

[image]

majianjia, 2021-07-17:
Thanks, 
It looks like there is an unsupported layer with the sequential model.