majianjia/nnom #139: KWS Keras Model Accuracy

HadeelMabrouk, 2021-07-31:
Hi,

I was wondering what is the reported accuracy for the Keras model in the KWS example. [Here](https://github.com/majianjia/nnom/tree/master/examples/keyword_spotting), the reported top-1 acc is reported to be = 83.371195%. However, in [here](https://github.com/majianjia/nnom/tree/master/examples) it's reported to be around 90%. So, I was wondering if there was a typo in this part.

On the other hand, when I tried to retrain the Keras model (using the old configurations), I only got an accuracy of 81.7%. So, I was wondering what might be the reason for these inconsistencies.

TIA.

majianjia, 2021-07-31:
Hi,
The training set might be different from time to time (such as volume, noise mixing level, epoch, selected keywords, or the model itself). It might reach around 90% back in time.
The accuracy for all 35 keywords will be low, but will increase if you only select 10. 
You can tune the model once you understand the process.

HadeelMabrouk, 2021-08-08:
Noted with thanks.

closed 2021-08-08 by HadeelMabrouk