majianjia/nnom #142: what does the input data looks like when I wanna generate weight.h for a NN that includes Concatenate layer?

Dark-Chao, 2021-09-24:
hi,
I have my NN with Concatenate layer for 3 branches. My input data for each branch is nparray(2974, 128, 3), where 2974 is the number of data pieces. 
When I want to generate weight.h by using **nnom.generate_model(model, data, name='weights.h')**, I have no idea how the **data** required by the function looks like. Is it a list of 3 nparray? or what type and shape it should be?
[image]

Many Thanks
Chao

majianjia, 2021-09-24:
The current version doesn't support multiple inputs. 
The input data is only to calibrate the model, i.e. run the model and get the intermediate activation ranges for quantisation. 
It can be a segment of your training/testing data, same shape but with a different number of pieces. 

If you want to use a multiple inputs model as you showed.
- In the current version, you can split the 3 branches into individual models, then use concatenate for their output. 
- You might also concate the 3 input to one pieces then use 3x cropping with different setting to split the data.

Dark-Chao, 2021-09-26:
Thanks for your reply.
Are you gonna upgrade it to support this feature in the future?

majianjia, 2021-09-26:
yes i will add this function

Dark-Chao, 2021-09-26:
Please let me know when it's available. 

Thanks!!