majianjia/nnom #144: Code error for 'main.c' when running the auto test example with my own dataset

siddhant0718, 2021-11-15:
As I wanted to check the feasibility of the NNOM for the MSP430 - I used the auto test example but changed the dataset used in the code - instead of mnist, I use the 'iris' dataset and also change the network to a very small configuration - the python scripts runs correctly and generates a 'weights.h' file which is compatible in size to the requirements of the MCU. The 'main.c' file is the same as that for the example as I am only changing the inputs in the code and all changes associated with the same are already incorporated when the 'weights.h' file would be called. However, after debugging the code for syntax and trying to load it on the MCU - I get the following error:
[image]

I am not sure exacty what is going wrong with the code as generally assert() throws an error if the code is buggy but from what I have read and tried to work out for my case - there have not been any changes in the code that should trigger this result. I am sure I might be missing something here but if someone could point me in a direction as to how I can resolve this, it would be highly appreciated.
Also attaching the main.c file for reference.
[main.txt](https://github.com/majianjia/nnom/files/7538247/main.txt)

Thanks!

majianjia, 2021-11-15:
It looks like your program cannot find the data file. So it shows an assert fault when validating if the file is open with "read".  
You should copy your dataset file to the same folder or modify the path to your data file.

siddhant0718, 2021-11-15:
Oh yes, thanks. I just checked and it seems like the file got copied into a different location.

closed 2021-11-15 by siddhant0718