majianjia/nnom #148: Running the NNOM on MSP430 - layers not processed

siddhant0718, 2021-12-06:
I am trying to use the NNOM on MSP430 MCU - when executing the auto test and keyword spotting examples with their respective 'main.c' files using the 'weights.h' files generated respectively, I am encountering a common error for both cases. I have a small network of Fully Connected (Dense) layers. When creating the model using **nnom_model_create()** , the input layer is correctly processed and I can see the status for the same in my debugger. However, when I move to the dense layer, the function returns a NULL for the layer. After debugging the code, I see that - **void *nnom_mem(size_t size)** fails to assign the size to the pointer p defined in the code and instead makes it zero - in turn this trigger the layer==NULL condition in the **nnom_dense.c** and the model fails to predict correctly as all other layers which are also of the 'dense' format fail to be initialized. I cannot figure out why this does not work. The only change I make is to the 'nnom_port.h' file where I initialize the '#define NNOM_USING_STATIC_MEMORY' - as the input layer also fails in case this defintion is commented out. Could you please suggest what changes I need to make and also what I might be observing wrong if that is the case, as this would be extremely helpful? Please let me know if any other details are necessary for clarification. 
[nnom.h.txt](https://github.com/majianjia/nnom/files/7659440/nnom.h.txt)

majianjia, 2021-12-06:
Before you `#define NNOM_USING_STATIC_MEMORY` It is probably related to not having enough stack memory. 

After you `#define NNOM_USING_STATIC_MEMORY` remember to set it first using `nnom_set_static_buf()` before you call `nnom_model_create()`.
You can see here for an example

https://github.com/majianjia/nnom/blob/44c3cfed7e74ee32dd1b1f1e4cdaf49ad1cf2def/examples/auto_test/main.c#L71

Make sure you assigned enough memory to it.   If you are not sure how much buffer you need, you can run your model on PC and see the log for memory size required. It is recommend to add a bit more incase of platform difference.

siddhant0718, 2021-12-06:
Thanks for the quick reply, I had set the static memory but in the wrong file - after defining it in the correct location, the layers are now detected.

closed 2021-12-06 by siddhant0718