majianjia/nnom #150: CEVA x2动态空间问题。

lzlwakeup, 2021-12-10:
PC端VS2019可以运行成功KWS。
移植Ceva dsp的过程中，出现内存申请无法运行问题。
例如计算mfcc中，若在create中申请，则编译烧录Ceva后，出现mfcc成员数据打印错误，即数值飞掉。将mfcc中成员指针改为具体大小的数组，则可以正确输出。

model = nnom_model_create();
输出打印m->head null的错误，查找了一下问题，指向nnom_input.c中的input_s()函数，其他需要申请空间的位置应该都有该问题。
layer = nnom_mem(sizeof(nnom_io_layer_t) + sizeof(nnom_layer_io_t) * 2);
if (layer == NULL)
{
    printf("input err \n");
}
编译烧录打印，串口查看input err。其他情况相同。
我想请教，这种问题是否有解决方法或者解决思路？

majianjia, 2021-12-10:
nnom 需要比较大量的内存。如果默认stack只有几kB会遇到内存无法分配的情况。
解决方法是在port.h里面打开静态内存支持，然后使用数组来分配内存
参考这个
https://github.com/majianjia/nnom/blob/44c3cfed7e74ee32dd1b1f1e4cdaf49ad1cf2def/examples/auto_test/main.c#L69

如果排除这个问题，可能是内存对齐的问题。DSP可能对内存对齐有自己的要求。可以修改nnom.h头文件内的对齐大小宏。

lzlwakeup, 2021-12-13:
非常感谢！采用静态内存后，代码可以在dsp上进行初步的运行预测。