majianjia/nnom #152: 无法生成weight.h文件

qiansui-maker, 2021-12-21:
Model: "sequential_1"
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
lstm_1 (LSTM)                (None, 240, 64)           17664     
_________________________________________________________________
gru_1 (GRU)                  (None, 240, 64)           24768     
_________________________________________________________________
gru_2 (GRU)                  (None, 32)                9312      
_________________________________________________________________
dense_1 (Dense)              (None, 1)                 33        
=================================================================
Total params: 51,777
Trainable params: 51,777
Non-trainable params: 0

lstm_1_input_1:0 max value: 3.9511321146836504 min value: -34.524399790141466 dec bit 1
KLD loss [9.242177959557377, 0.16955467682553751, 0.12965383041279469, 0.11479619833139001]
KLD shift [1, 2, 3, 4]
lstm_1 is using KLD method, original shift 1 KLD results 4
lstm_1 max value: 3.9511321146836504 min value: -34.524399790141466 dec bit 4
KLD loss [9.242177959557377, 0.16955467682553751, 0.12965383041279469, 0.11479619833139001]
KLD shift [1, 2, 3, 4]
gru_1 is using KLD method, original shift 1 KLD results 4
gru_1 max value: 3.9511321146836504 min value: -34.524399790141466 dec bit 4
KLD loss [9.242177959557377, 0.16955467682553751, 0.12965383041279469, 0.11479619833139001]
KLD shift [1, 2, 3, 4]
gru_2 is using KLD method, original shift 1 KLD results 4
gru_2 max value: 3.9511321146836504 min value: -34.524399790141466 dec bit 4
dense_1 max value: 2.2803915 min value: -1.4103148 dec bit 5
shift list {'lstm_1_input_1': 1, 'lstm_1': 4, 'gru_1': 4, 'gru_2': 4, 'dense_1': 5}
weights for layer lstm_1
  weight: lstm_1_1/kernel:0
  original shape:  (4, 256)
  dec bit 7
  reshape to: (256, 4)
  weight: lstm_1_1/recurrent_kernel:0
  original shape:  (4, 256)
  dec bit 7
  reshape to: (256, 4)
  bias:  lstm_1_1/bias:0
  original shape:  (64, 256)
  dec bit 8
  reshape to: (256, 64)
weights for layer gru_1
  weight: gru_1_1/kernel:0
  original shape:  (64, 192)
  dec bit 7
  reshape to: (192, 64)
  weight: gru_1_1/recurrent_kernel:0
  original shape:  (64, 192)
  dec bit 7
  reshape to: (192, 64)
  bias:  gru_1_1/bias:0
  original shape:  (64, 192)
  dec bit 8
  reshape to: (192, 64)
weights for layer gru_2
  weight: gru_2_1/kernel:0
  original shape:  (64, 96)
  dec bit 7
  reshape to: (96, 64)
  weight: gru_2_1/recurrent_kernel:0
  original shape:  (64, 96)
  dec bit 7
  reshape to: (96, 64)
  bias:  gru_2_1/bias:0
  original shape:  (32, 96)
  dec bit 8
  reshape to: (96, 32)
weights for layer dense_1
  weight: dense_1_1/kernel:0
  original shape:  (32, 1)
  dec bit 8
gru_2_1

---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-13-fc920cd91782> in <module>
      9 
     10 model = load_model('my_pred_model.h5')
---> 11 generate_model(model, np.vstack((X_train, X_test)), name="weights.h")

~/workspace/nnom_utils.py in generate_model(model, x_test, name, format, kld)
    448 def generate_model(model, x_test, name='weights.h', format='hwc', kld=True):
    449     shift_list = layers_output_ranges(model, x_test, kld)
--> 450     generate_weights(model, name=name, format=format, shift_list=shift_list)
    451     if(type(model.layers[0]) != InputLayer):
    452         L = [model.input] + model.layers

~/workspace/nnom_utils.py in generate_weights(model, name, format, shift_list)
    250                 inp = layer.input.name.replace(':','/').split('/')[0]
    251                 print(inp)
--> 252                 input_encoding = shift_list[inp]
    253                 if ("kernel" in var_name):
    254                     weight_dec_shift = dec_bits

KeyError: 'gru_2_1'

majianjia, 2021-12-21:
可能是因为之前的session没有清空，脚本没办法识别第二个模型实例的名字。

可以删掉并清空model之后再载入。
https://github.com/majianjia/nnom/blob/44c3cfed7e74ee32dd1b1f1e4cdaf49ad1cf2def/examples/auto_test/main.py#L112
参考这几行。

qiansui-maker, 2021-12-22:
出现了新的错误
AttributeError: 'LSTM' object has no attribute 'inbound_nodes'

majianjia, 2021-12-22:
TF版本是啥

qiansui-maker, 2022-01-06:
> TF版本是啥

解决了，是版本问题

closed 2022-01-06 by qiansui-maker