majianjia/nnom #160: CMSIS-NN 使用加速問題

rxin-1219, 2022-05-04:
你好

使用CMSIS-NN 後端進行加速
我是在rt-thread 上進行
直接使用上面的CMSIS Package
[image]
有添加兩個 ' ARM_MATH_CM7, __FPU_PRESENT=1'  因為我是使用F746(不知道是不是使用ARM_MATH_CM7)
[image]
如果我有任何錯的地方請麻煩告訴我

非常感謝

majianjia, 2022-05-04:
你好，看提示应该是路径的问题，先保证你可以在工程内能使用到 arm_math.h

rxin-1219, 2022-05-05:
> 你好，看提示应该是路径的问题，先保证你可以在工程内能使用到 arm_math.h

不好意思 後來我成功編譯之後 發現這些錯誤
[image]
從上面看code 發現
[image]
是我使用了甚麼不支援的layer嗎?
抱歉耽誤你的寶貴時間

非常感謝

majianjia, 2022-05-05:
你好，看起来cmsis nn并没有加入到你的工程。你看看是否能在工程目录内找到cmsis nn相关的源文件?

rxin-1219, 2022-05-05:
後來我又把整個環境重建
確實把NN跟DSP 加進去 工程中
[image]