majianjia/nnom #169: h5

lzlwakeup, 2022-07-28:


lzlwakeup, 2022-08-02:
对于x_test的问题已经初步解决，生成了一个随机整数的ndarray，但是转换过程中报错了。

quantisation list {'image_input': [0, 0], 'zero_padding2d': [0, 0], 'conv2d': [-5, 0], 'batch_normalization': [-5, 0], 're_lu': [-5, 0], 'conv2d_1': [-5, 0], 'batch_normalization_1': [-5, 0], 're_lu_1': [-5, 0], 'zero_padding2d_1': [-5, 0], 'depthwise_conv2d': [-4, 0], 'batch_normalization_2': [-4, 0], 're_lu_2': [-4, 0], 'conv2d_2': [-4, 0], 'batch_normalization_3': [-4, 0], 'zero_padding2d_2': [-4, 0], 'conv2d_3': [-3, 0], 'batch_normalization_4': [-3, 0], 're_lu_3': [-3, 0], 'conv2d_4': [-3, 0], 'batch_normalization_5': [-3, 0], 're_lu_4': [-3, 0], 'depthwise_conv2d_1': [-4, 0], 'batch_normalization_6': [-4, 0], 're_lu_5': [-4, 0], 'conv2d_5': [-3, 0], 'batch_normalization_7': [-3, 0], 'concatenate': [-3, 0], 'conv2d_6': [-3, 0], 'batch_normalization_8': [-3, 0], 're_lu_6': [-3, 0], 'zero_padding2d_3': [-3, 0], 'depthwise_conv2d_2': [-3, 0], 'batch_normalization_9': [-3, 0], 're_lu_7': [-3, 0], 'conv2d_7': [-1, 0], 'batch_normalization_10': [-1, 0], 'conv2d_8': [-2, 0], 'batch_normalization_11': [-2, 0], 're_lu_8': [-2, 0], 'depthwise_conv2d_3': [-2, 0], 'batch_normalization_12': [-2, 0], 're_lu_9': [-2, 0], 'conv2d_9': [-1, 0], 'batch_normalization_13': [-1, 0], 'dropout': [-1, 0], 'add': [-1, 0], 'conv2d_10': [-1, 0], 'batch_normalization_14': [-1, 0], 're_lu_10': [-1, 0], 'depthwise_conv2d_4': [-2, 0], 'batch_normalization_15': [-2, 0], 're_lu_11': [-2, 0], 'conv2d_11': [-1, 0], 'batch_normalization_16': [-1, 0], 'dropout_1': [-1, 0], 'add_1': [-2, 0], 'conv2d_12': [-1, 0], 'batch_normalization_17': [-1, 0], 're_lu_12': [-1, 0], 'depthwise_conv2d_5': [-2, 0], 'batch_normalization_18': [-2, 0], 're_lu_13': [-2, 0], 'conv2d_13': [-2, 0], 'batch_normalization_19': [-2, 0], 'dropout_2': [-2, 0], 'add_2': [-3, 0], 'concatenate_1': [-3, 0], 'conv2d_14': [-2, 0], 'batch_normalization_20': [-2, 0], 're_lu_14': [-2, 0], 'zero_padding2d_4': [-2, 0], 'depthwise_conv2d_6': [-2, 0], 'batch_normalization_21': [-2, 0], 're_lu_15': [-2, 0], 'conv2d_15': [-1, 0], 'batch_normalization_22': [-1, 0], 'conv2d_16': [0, 0], 'batch_normalization_23': [0, 0], 'conv2d_17': [-1, 0]}
fusing batch normalization to conv2d

Traceback (most recent call last):
  File "con.py", line 23, in <module>
    generate_model(model, x_test[:100], name='weights.h')
  File "/home/lzl/Documents/nnom/scripts/nnom.py", line 748, in generate_model
    quantize_weights(model, per_channel_quant=per_channel_quant, name=name, format=format, layer_q_list=layer_q_list)
  File "/home/lzl/Documents/nnom/scripts/nnom.py", line 606, in quantize_weights
    fuse_bn_to_conv(layer)
  File "/home/lzl/Documents/nnom/scripts/nnom.py", line 41, in fuse_bn_to_conv
    c_b = layer.get_weights()[1]
IndexError: list index out of range
这个超过范围没搞明白是哪里超过

lzlwakeup, 2022-08-02:
conv2d_13 Quantized method: max-min  Values max: 118.52757 min: -150.47974 dec bit -1
4/4 [==============================] - 0s 7ms/step
batch_normalization_19 Quantized method: max-min  Values max: 305.4049 min: -307.9548 dec bit -2
dropout_2 Quantized method: max-min  Values max: 305.4049 min: -307.9548 dec bit -2
4/4 [==============================] - 0s 8ms/step
add_2 Quantized method: max-min  Values max: 407.15002 min: -351.8012 dec bit -2
concatenate_1 Quantized method: max-min  Values max: 407.15002 min: -351.8012 dec bit -2
4/4 [==============================] - 0s 7ms/step
conv2d_14 Quantized method: max-min  Values max: 353.57272 min: -359.15775 dec bit -2
4/4 [==============================] - 0s 8ms/step
batch_normalization_20 Quantized method: max-min  Values max: 184.71503 min: -282.97308 dec bit -2
re_lu_14 Quantized method: max-min  Values max: 184.71503 min: -282.97308 dec bit -2
zero_padding2d_4 Quantized method: max-min  Values max: 184.71503 min: -282.97308 dec bit -2
4/4 [==============================] - 0s 8ms/step
depthwise_conv2d_6 Quantized method: max-min  Values max: 104.83797 min: -115.963394 dec bit 0
4/4 [==============================] - 0s 8ms/step
batch_normalization_21 Quantized method: max-min  Values max: 438.39203 min: -280.17523 dec bit -2
re_lu_15 Quantized method: max-min  Values max: 438.39203 min: -280.17523 dec bit -2
4/4 [==============================] - 0s 8ms/step
conv2d_15 Quantized method: max-min  Values max: 169.05531 min: -163.94919 dec bit -1
4/4 [==============================] - 0s 8ms/step
batch_normalization_22 Quantized method: max-min  Values max: 199.35162 min: -237.11674 dec bit -1
4/4 [==============================] - 0s 9ms/step
conv2d_16 Quantized method: max-min  Values max: 138.25987 min: -154.78223 dec bit -1
4/4 [==============================] - 0s 8ms/step
batch_normalization_23 Quantized method: max-min  Values max: 125.73436 min: -119.78797 dec bit 0
4/4 [==============================] - 0s 9ms/step
conv2d_17 Quantized method: max-min  Values max: 41.096416 min: -248.74689 dec bit -1
set dec bit -3 for the input of concatenate_1 : ['add_2', 're_lu_7']
set dec bit -2 for the input of add_2 : ['add_1', 'dropout_2']
set dec bit -1 for the input of add_1 : ['add', 'dropout_1']
set dec bit -1 for the input of add : ['batch_normalization_10', 'dropout']
set dec bit -3 for the input of concatenate : ['batch_normalization_7', 're_lu_3']

这是前面的部分打印。

lzlwakeup, 2022-08-02:
细看了一下，是要将conv和bn层进行融合，过程中报错。模型用的是
https://github.com/EdgeAIWithRTT/Project4-Person_detection_RT-AK/tree/main/model/yolo-s_with_relu
里面的yolo-s_with_relu.h5的这个模型

lzlwakeup, 2022-08-03:
issue更新：
前面说的问题已经确认，确实是conv2d层不带bias引起转换错误。
更改方式为：
1、注释掉fuse过程，在未更新工程前，不添加编译融合。
 # try to fuse BN layer to convolutional
        '''
        if ('conv' in layer.name) and \
            ('batch_normalization' in layer.outbound_nodes[0].outbound_layer.name):
            fuse_bn_to_conv(layer)
            '''
 # generate weights and bias now
2、对于有bias的conv2d层，单独进行判断，给bias，其余无bias的conv2d层，不添加转换
def gen_conv2d_config(layer, output_shifts, bias_shifts):
   ...
    c = c.replace('<weight>',convert_tensor_name(layer.weights[0]))
    if (layer.name == 'conv2d_17'):
        c = c.replace('<bias>',convert_tensor_name(layer.weights[1]))
    c = c.replace('<output_shift_values>', output_shifts)
   ...
    return c
这样转换可以完成，但是有个疑问，gen_conv2d_trans_config也包含bias转换，但是未报错，不知道具体使用会有和影响。先编译过去，搭建测试文件试一下。

lzlwakeup, 2022-08-03:
issue更新：
模型转换完毕，但是bias的问题还是存在。前面说注释掉fuse过程，conv2d和bn单独转换，看起来好象是不行，部分报错如下：
weights.h:848:55: error: ‘CONV2D_11_OUTPUT_RSHIFT’ undeclared here (not in a function); did you mean ‘CONV2D_17_OUTPUT_RSHIFT’?
  848 | const nnom_qformat_param_t conv2d_11_output_shift[] = CONV2D_11_OUTPUT_RSHIFT;
      |                                                       ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                       CONV2D_17_OUTPUT_RSHIFT
weights.h:849:53: error: ‘CONV2D_11_BIAS_LSHIFT’ undeclared here (not in a function); did you mean ‘CONV2D_17_BIAS_LSHIFT’?
  849 | const nnom_qformat_param_t conv2d_11_bias_shift[] = CONV2D_11_BIAS_LSHIFT;
      |                                                     ^~~~~~~~~~~~~~~~~~~~~
      |                                                     CONV2D_17_BIAS_LSHIFT
weights.h:854:30: error: expected expression before ‘<’ token
  854 |     .bias = (nnom_tensor_t*)&<bias>,
      |                              ^
weights.h:884:55: error: ‘CONV2D_12_OUTPUT_RSHIFT’ undeclared here (not in a function); did you mean ‘CONV2D_17_OUTPUT_RSHIFT’?
  884 | const nnom_qformat_param_t conv2d_12_output_shift[] = CONV2D_12_OUTPUT_RSHIFT;
      |                                                       ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                       CONV2D_17_OUTPUT_RSHIFT
weights.h:885:53: error: ‘CONV2D_12_BIAS_LSHIFT’ undeclared here (not in a function); did you mean ‘CONV2D_17_BIAS_LSHIFT’?
  885 | const nnom_qformat_param_t conv2d_12_bias_shift[] = CONV2D_12_BIAS_LSHIFT;
      |                                                     ^~~~~~~~~~~~~~~~~~~~~
      |                                                     CONV2D_17_BIAS_LSHIFT
目前看起来是，bias引起RSHIFT和LSHIFT命名错误。

majianjia, 2022-08-04:
量化需要少量实际数据来校准。不支持bias=false的模式

lzlwakeup, 2022-08-04:
另一个问题，换了模型，提示bias在使用前未定义
Traceback (most recent call last):
  File "con.py", line 23, in <module>
    generate_model(model, x_test[:10], name='weights.h')
  File "../nnom/scripts/nnom.py", line 749, in generate_model
    quantize_weights(model, per_channel_quant=per_channel_quant, name=name, format=format, layer_q_list=layer_q_list)
  File "../nnom/scripts/nnom.py", line 732, in quantize_weights
    f.write('#define ' + layer.name.upper() + '_BIAS_LSHIFT '+ to_cstyle(bias_shift) +'\n\n')
UnboundLocalError: local variable 'bias_shift' referenced before assignment