majianjia/nnom #182: Is 0.4.3 NNOM c backend slower than CMSIS-5.9.0?

bfs18, 2023-03-02:
I test the mnist-simple model in rt-thread environment and run model predict for 1000 times.
using CMSIS 5.9.0 backend, it consumes 1423 ms
using NNOM 0.4.3 C backend, it consumes 1392 ms
The document says that replacing c backend with CMSIS would bring 5 times speed-up, however, the performance is similar according to my test. What's the reason? NNOM updated or I didn't set CMSIS macros correctly?

majianjia, 2023-03-03:
Probably you havent enable CMSIS's NN acceleration, i.e. you are running CMSIS on their C backend instead of SIMD assemblys.
You may check CMSIS instruction or [the note under HWC format](https://github.com/majianjia/nnom/blob/master/docs/Porting_and_Optimisation_Guide.md#format-configuration)

bfs18, 2023-03-04:
> 

Hi Jia, thanks for your advice. I add 
```
ARM_MATH_DSP,ARM_MATH_CM7,__FPU_PRESENT=1 
```
using CMSIS 5.9.0 backend, reduces to **1055** ms.
I used the rt-thread **qemu-vexpress-a9** bsp. Is this normal?