majianjia/nnom #183: Is v0.4.3 nnom.py support sigmoid?

songdaw, 2023-03-05:
Hi,
   My model use sigmoid activation like:
     activation = 'sigmoid'
     output = Activation(activation)(input)
and I use generate_model() in nnom.py to convert model.
I think should add 
     ('activation' in layer.name and layer.get_config()['activation'] == 'sigmoid') 
to function is_shift_layer() .

majianjia, 2023-03-06:
For sigmoid, please use Sigmoid layer instead of Activation("sigmoid") at the moment

songdaw, 2023-03-07:
Sorry, I can't find how to use Sigmoid layer, but tf.keras.activations.sigmoid() which produce Lambda layer. Did I miss something?
I use keras=2.6.0.

jonnor, 2024-03-30:
Is this the right fix, just to add sigmoid to is_shift_layer ? Or are there other complications?

Btw, a workaround might be to use hard_sigmoid, it appears to be implemented

songdaw, 2024-04-01:
sigmoid is already implemented, try replace act_hard_sigmoid() with act_sigmoid().

---- Replied Message ----
| From | Jon ***@***.***> |
| Date | 03/30/2024 11:20 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [majianjia/nnom] Is v0.4.3 nnom.py support sigmoid? (Issue #183) |

Is this the right fix, just to add sigmoid to is_shift_layer ? Or are there other complications?

Btw, a workaround might be to use hard_sigmoid, it appears to be implemented