majianjia/nnom #199: about reshape

panhu, 2023-05-26:
Hi:
I want to ask what the nnom can support tf.reshape ?

majianjia, 2023-05-26:
Could you check if this implementation is working?
https://github.com/majianjia/nnom/blob/master/src/layers/nnom_reshape.c

panhu, 2023-05-26:
Hi:
I used a instead of b in my model, and when generating the. h file, it gave me "model. hook (reshappe_s (&tf. reshappe_12_config), layer [1];", but there was no generat the other definitions of "tf. reshape_ 12_config" .
and i used was "tf.reshape" not "tf.keras.layers.Reshape".

panhu, 2023-05-30:
Hi:
I used "tf.reshape" instead of  "tf.keras.layers.Reshape" in my model, and when generating the. h file, it gave me "model. hook (reshappe_s (&tf. reshappe_12_config), layer [1];", but there was no generat the other definitions of "tf. reshape_ 12_config" .
and i used was "tf.reshape" not "tf.keras.layers.Reshape".