majianjia/nnom #201: 关于输出维度的问题

panhu, 2023-06-05:
您好：
我想请问下假设我的输出是（1,256,2）也就是512个数。那么输出是什么样的，我看输出好像直接就是512个数了，那么它对于(1,256,2)的组织形式是啥样的，怎么区分或者找出这两个256呢？

majianjia, 2023-06-06:
默认情况下数据是HWC数据结构，配置时恒定HWC格式（你看到的shape）
如果这两者匹配，C语言下使用 `data[1][256][2]` 访问你的数据

panhu, 2023-06-06:
感谢您的回复，我看对于给定输出是tensor_output_dim = {1,256,2}，输出的nnom_output_data好像是一维的数组512，它们的对应关系是什么样的，是前256为[0][256][0]维与后256为[0][256][1]组成还是交叉组成512的一维数组

majianjia, 2023-06-07:
交叉的
实际排列方式为`data[1][256][2]`

panhu, 2023-06-07:
好的，感谢您的回复。遇到一个奇怪的问题，在PC上我是能够运行add_s与concat_s层的。但是芯片烧录后却是跑到add_s与concat_s时就不动了。也不报错 好像卡死了一样，很奇怪

panhu, 2023-06-07:
好的，感谢您的回复。遇到一个奇怪的问题，在PC上我是能够运行add_s与concat_s层的。但是芯片烧录后却是跑到add_s或concat_s时就不动了。也不报错 好像卡死了一样，很奇怪，然后我又在matrix.c中每一个步骤进行测试好像都是正常运行的，但返回后程序就跑飞了

panhu, 2023-12-11:
谢谢，假设我有输入数据为实部128，虚部128，那我是不是可以之间将它们放在一起组成[256]个数组作为输入呢，是之间实部+虚部进行组合，还是应该实虚实虚这样交叉放入