majianjia/nnom #214: Keras 3 version compatibility

jonnor, 2024-03-24:
Hi. I cannot see that there is any documentation on which Keras versions are supported. I tried running with the latest versions as per today - Keras 3.1.1. 

This gives the following exception.

```
Traceback (most recent call last):
  File "/home/jon/projects/embeddedml/handson/nnom/train_mobilenet.py", line 27, in <module>
    nnom.generate_model(model, data, name=out_path)
  File "/home/jon/projects/embeddedml/handson/nnom/nnom/scripts/nnom.py", line 757, in generate_model
    quantize_weights(model, per_channel_quant=per_channel_quant, name=name, format=format, layer_q_list=layer_q_list)
  File "/home/jon/projects/embeddedml/handson/nnom/nnom/scripts/nnom.py", line 610, in quantize_weights
    print(dir(layer.outbound_nodes[0].output_tensors))
              ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Conv2D' object has no attribute 'outbound_nodes'. Did you mean: '_outbound_nodes'?
```

I tried to fix that line, but then just get stuck a couple of lines down.

```
Traceback (most recent call last):
  File "/home/jon/projects/embeddedml/handson/nnom/train_mobilenet.py", line 27, in <module>
    nnom.generate_model(model, data, name=out_path)
  File "/home/jon/projects/embeddedml/handson/nnom/nnom/scripts/nnom.py", line 757, in generate_model
    quantize_weights(model, per_channel_quant=per_channel_quant, name=name, format=format, layer_q_list=layer_q_list)
  File "/home/jon/projects/embeddedml/handson/nnom/nnom/scripts/nnom.py", line 614, in quantize_weights
    ('batch_normalization' in layer.outbound_nodes[0].outbound_layer.name):
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'Node' object has no attribute 'outbound_layer'
```

So I believe Keras 3.x does not really work. But which version should I go back to that is known to be working?

jonnor, 2024-04-07:
For now TF 2.12.1 and TF 2.14.1 with Keras 2.x works. Keras 3 is getting more popular though, so this issue should probably stay open to track progress in supporting that.

kanndil, 2024-08-22:
Make sure to have Python3.11 or less installed, since TF 2.12.1 and TF 2.14.1 are not compatible with Python3.12

songdaw, 2024-10-29:
Keras 2.6.0 works fine for me, I think what we really need is the model weights. So, I use Pytorch to debug and train the model, then transfer model weights to Keras, at last to NNOM.